KlipperScreen/KlipperScreen #952: [BUG] Screen shows console instead of KlipperScreen

Roccccky, 2023-04-17:
I set up a fresh installation of fluidd and don't get klipperscreen to show up. 

I followed all steps through kiauh. Installed bullseye lite with raspy-imager (ssh/wlan/etc) and went through all steps in kiauh (Klipper/Moonraker/Fuidd/Sklippersceen). Klipper and Printer is running but screen shows console instead of klipperscreen. Tried all solutions from troubleshooting guide without success. Also tried buster lite and desktop verison of raspberry os. still no klipperscreen.

I had an old installation running before so i'm 100% sure hardware and drivers are working. something has changed for loginconsole so i stuck in shows console and can get klipperscreen to run :(

Arlandw, 2023-04-18:
I have this same issue - was working and then stopped

Aurescalia, 2023-04-30:
Hello, i have kind of the same problem with a 7inch raspberry official screen : i had klipper screen working but no touch so i went through troubleshooting steps and since i entered GL menu it downloads its update and klipper screen has gone and i can’t display it again even changing fb0 / fb1 stuff… 😩

alfrix, 2023-05-14:
same as #971  remove fbturbo
not a bug of ks, it's an issue with the OS and the installed screen of the users

closed 2023-05-14 by alfrix

Z4nterox, 2023-05-14:
This would require that you have fturbo installed in the first place.

Roccccky, 2023-06-25:
I found a solution for this. I swapped fbturbo and use fbdev .. worked for me

`#remove fbturbo and use fbdev

sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf

sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
#Add this text ans save
Section "Device"
	Identifier "LCD"
	Driver "fbdev"
	Option "fbdev" "/dev/fb1"
	Option "Rotate"  "180"
	Option "SwapbuffersWait" "true"
EndSection`

i also had to flip my touchinput

`#flip touchinput 
sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf
 Section "InputClass"

        Identifier "libinput touchscreen catchall"

        MatchIsTouchscreen "on"

        MatchDevicePath "/dev/input/event*"

        Driver "libinput"

        Option "TransformationMatrix" "-1 0 1 1 1 0 0 0 1"

EndSection
`
also check this link
https://klipperscreen.readthedocs.io/en/latest/Hardware/#touchscreen-touch-rotation

Roccccky, 2023-06-25:
I found a solution for this. I swapped fbturbo and use fbdev .. worked for me

###remove fbturbo and use fbdev

sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf`

sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf`

####Add this text ans save

Section "Device"
	Identifier "LCD"
	Driver "fbdev"
	Option "fbdev" "/dev/fb1"
	Option "Rotate"  "180"
	Option "SwapbuffersWait" "true"
EndSection

i also had to flip my touchinput

###flip touchinput 

sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf

Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "-1 0 1 1 1 0 0 0 1"
EndSection

also check this link
https://klipperscreen.readthedocs.io/en/latest/Hardware/#touchscreen-touch-rotation