KlipperScreen/KlipperScreen #954: [BUG (sort of)] Some documentation improvements

tcurdt, 2023-04-22:
**Describe the bug**

I feel a bit let down by the documentation :)

First off - fantastic project. I got it working just fine, but got stuck when configuration some aspects.

Now before I open tickets for the various aspects I thought I ask first what I might be missing.
I tried to reach out on discord but so far haven't gotten any responses.

Here is a list of things that I couldn't find in the docs yet:

- After a while during a print I noticed that the screen got "stuck". No black screen. Just not updated. As it turns out that's the screensaver (that seems to have a problem with my setup). But I couldn't find any information [in the docs ](https://klipperscreen.readthedocs.io/en/latest/Configuration/) on how to configure it.
- Neither the configuration section nor the [section on themes](https://klipperscreen.readthedocs.io/en/latest/Theming/) seems to cover how to actually select a theme.
- The size of the back button is a problem (for me). It's too small. It's quite easy to touch e.g. the "Home All" (Argh!). Now I guess that's more of an issue of the theme. But I am also a bit lost on which classes are what. Since there is not HTML or browser - any advice on how iterate on this?

I happy to split this up into individual tickets if you prefer - just thought I check first.

alfrix, 2023-05-14:
hw is different if your screen don't properly support dpms to turn off disable dpms in the settings it will stay on but black

https://klipperscreen.readthedocs.io/en/latest/Panels/#settings
select a theme in the settings link above
if size is a problem you can tweak it using themes, explaining code is something that most likely will never happen, since it is more time consuming than writing the code itself, and i'm having little spare time, if you want to contribute to a pr

closed 2023-05-14 by alfrix

tcurdt, 2023-05-14:
> select a theme in the settings link above

The question is about creating a theme - not switching it.

> if size is a problem you can tweak it using themes, 

That's my intention.

> explaining code is something that most likely will never happen, since it is more time consuming than writing the code itself, and i'm having little spare time, if you want to contribute to a pr

I am not asking to give some guidance on where to start.

Of course there is the css here https://github.com/jordanruthe/KlipperScreen/tree/master/styles

But how do I get to the see the HTML to know which elements to style?

I can just copy a theme, switch to using it and then change the CSS while klipperscreen is running.
But after all there is no browser to hit reload on. Restarting klipperscreen after every change?

If would be nice to document on a high level how to do this. Not about "explaining code".