KlipperScreen/KlipperScreen #956: [BUG] Either no touch function or only console at boot

z28chris, 2023-04-22:
I am using this screen: https://www.waveshare.com/wiki/4.3inch_DSI_LCD?Amazon#Software_setting
After adding `dtoverlay=vc4-kms-v3d` and `dtoverlay=vc4-kms-dsi-7inch` to the config.txt (as prompted on mentioned website), I installed KlipperScreen via kiauh. It then starts KlipperScreen fine, but the touch wouldn't work. So I changed the "kms" to "fkms" as advised on the KlipperScreen troubleshooting section. Now It will only boot to console and KlipperScreen won't start at all. I checked the troubleshooting section for this, but with no success (`ls /dev/fb*` will only show me one entry fb0). What else can I do to get this running?

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11302433/KlipperScreen.log)

z28chris, 2023-04-22:
Sorry, it seems the log upload didn't work. So here's the log:

--------------------KlipperScreen Log Start--------------------
Git Version: v0.3.2-22-g282799c
2023-04-22 22:13:16,188 [screen.py:main()] - KlipperScreen version: v0.3.2-22-g282799c
2023-04-22 22:13:16,437 [config.py:get_config_file_location()] - Passed config (-c): /home/pi/KlipperScreen.conf
2023-04-22 22:13:16,471 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/defaults.conf
2023-04-22 22:13:16,493 [config.py:__init__()] - Configuration validated succesfuly
2023-04-22 22:13:16,494 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-04-22 22:13:16,595 [config.py:create_translations()] - Selected lang: None OS lang: en_GB
2023-04-22 22:13:16,596 [config.py:install_language()] - Using system lang
2023-04-22 22:13:16,598 [config.py:install_language()] - Using lang en
2023-04-22 22:13:16,613 [screen.py:__init__()] - Screen resolution: 800x480
2023-04-22 22:13:16,614 [KlippyGtk.py:__init__()] - Font size: 16.6 (medium)
2023-04-22 22:13:17,531 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2023-04-22 22:13:17,543 [screen.py:set_screenblanking_timeout()] - Using DPMS
2023-04-22 22:13:17,559 [screen.py:initial_connection()] - Default printer: None
2023-04-22 22:13:17,560 [screen.py:connect_printer()] - Connecting to printer: Printer
2023-04-22 22:13:17,560 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-04-22 22:13:17,561 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2023-04-22 22:13:17,640 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-04-22 22:13:17,675 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-04-22 22:13:17,693 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2023-04-22 22:13:17,694 [KlippyWebsocket.py:connect()] - Starting websocket thread
2023-04-22 22:13:17,706 [_logging.py:info()] - Websocket connected
2023-04-22 22:13:17,706 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2023-04-22 22:13:17,738 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-04-22 22:13:17,780 [KlippyRest.py:send_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/machine/device_power/devices
2023-04-22 22:13:17,800 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2023-04-22 22:13:17,823 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2023-04-22 22:13:17,825 [printer.py:reinit()] - Klipper version: v0.11.0-148-g52f4e20c
2023-04-22 22:13:17,826 [printer.py:reinit()] - # Extruders: 0
2023-04-22 22:13:17,826 [printer.py:reinit()] - # Temperature devices: 0
2023-04-22 22:13:17,826 [printer.py:reinit()] - # Fans: 0
2023-04-22 22:13:17,827 [printer.py:reinit()] - # Output pins: 0
2023-04-22 22:13:17,827 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2023-04-22 22:13:18,082 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks&motion_report&firmware_retraction&exclude_object
2023-04-22 22:13:18,083 [printer.py:change_state()] - Changing state from 'disconnected' to 'error'
2023-04-22 22:13:18,083 [printer.py:change_state()] - Adding callback for state: error
2023-04-22 22:13:18,098 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-04-22 22:13:18,099 [printer.py:init_temp_store()] - Temp store: []
2023-04-22 22:13:18,115 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-04-22 22:13:18,116 [screen.py:init_tempstore()] - Temperature store size: 1200
2023-04-22 22:13:18,117 [files.py:initialize()] - Gcodes path: None
2023-04-22 22:13:18,117 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2023-04-22 22:13:18,118 [screen.py:init_printer()] - Printer initialized
2023-04-22 22:13:18,150 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-04-22 22:13:20,111 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-04-22 22:13:20,112 [printer.py:init_temp_store()] - Temp store: []
2023-04-22 22:13:20,141 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-04-22 22:13:20,143 [screen.py:init_tempstore()] - Temperature store size: 1200
--------------------KlipperScreen Log Start--------------------
Git Version: v0.3.2-22-g282799c
2023-04-22 23:32:33,502 [screen.py:main()] - KlipperScreen version: v0.3.2-22-g282799c
2023-04-22 23:32:33,787 [config.py:get_config_file_location()] - Passed config (-c): /home/pi/KlipperScreen.conf
2023-04-22 23:32:33,789 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/defaults.conf
2023-04-22 23:32:33,811 [config.py:__init__()] - Configuration validated succesfuly
2023-04-22 23:32:33,812 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
[... 21 lines cut ...]
[... cut ...]
2023-04-22 23:32:35,764 [screen.py:init_printer()] - Printer initialized
2023-04-22 23:32:35,797 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-04-22 23:32:37,708 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-04-22 23:32:37,708 [printer.py:init_temp_store()] - Temp store: []
2023-04-22 23:32:37,737 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-04-22 23:32:37,738 [screen.py:init_tempstore()] - Temperature store size: 1200

Roccccky, 2023-04-23:
Same issue here :( ... reported last week.

closed 2023-04-25 by z28chris