KlipperScreen/KlipperScreen #963: [FR] Match shutdown action to Moonraker, Restart to Mainsail

tinymachines3d, 2023-04-29:
**Is your feature request related to a problem? Please describe.**
When using some non RPI SBCs, shutdown from KlipperScreen just reboots the device instead of shutting down. This is due to the shutdown method used. 

**Describe the solution you'd like**
KlipperScreen to use the method described in moonraker OR give you the option to select **halt** or default to poweroff
[image]

**Describe alternatives you've considered**
Only shutting down from mainsail 

**Additional context**
Is there another way to set this that is not otherwise documented?

alfrix, 2023-05-14:
while connected to moonraker the shutdown action is performed by moonraker and when selecting "printer" in the system panel shutdown/reboot options
otherwise ks will send `systemctl poweroff` and  ` systemctl reboot` if that doesn't work is a bug of the OS, i will not make patches to workaround broken platforms

closed 2023-05-14 by alfrix

dotorg, 2024-04-01:
For what its worth, not all Linux systems use systemd. _systemctl poweroff_ and _systemctl reboot_ not working isn't an OS bug. The low-level standard Linux command that works on all platforms is the shutdown command, which works on systems that don't use systemd as well. If you wanted to ensure compatibility, that's the command to use -- _shutdown -h_ or _shutdown -r_ 

A significant percentage of distributions targeted at minimal installations don't use it.