KlipperScreen/KlipperScreen #971: Updated to v0.3.2-27-g0b973ad via Mainsail updated manager - KlipperScreen no longer works

rosspirtle, 2023-05-07:
**Updated to v0.3.2-27-g0b973ad via Mainsail updated manager - KlipperScreen no longer works**

I used the update manager of Mainsail and rebooted the printer.  The screen initially looks like it is starting properly (startup log items scroll by as usual). When startup finishes, the screen shows only a non-blinking cursor in the top-left of the screen.

I tried un-installing and re-installing KlipperScreen and ended up with the same issue.  Used Uninstall.sh and then removed the KlipperScreen directory.  Then re-installed using git clone and KlipperScreen-install.sh.

The last three lines in the attached log file shows what appears to be the final steps of the update.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11415875/KlipperScreen.log)

dimaiv7, 2023-05-07:
me too

klipper
v0.11.0-198-g33b18fd6
KlipperScreen
v0.3.2-27-g0b973ad
moonraker
v0.8.0-40-gb21f177
mainsail
v2.5.1 

 OS: Raspbian GNU/Linux 10 (buster)
Distribution: MainsailOS 0.6.1 (buster)

jkaterenchuk, 2023-05-08:
Same here also.  Verison installed the same as @dimaiv7.

derDoc81, 2023-05-08:
Also same here

P3ps, 2023-05-08:
Hi,

I have exactly the same problem.

klipper
v0.11.0-198-g33b18fd6
KlipperScreen
v0.3.2-27-g0b973ad
mainsail
v2.5.1
moonraker
v0.8.0-40-gb21f177
crowsnest
v3.0.7-3-g20ed6a8
sonar
v0.1.2-0-g7962a12
timelapse
v0.0.1-132-gf9198e0

OS: Raspbian GNU/Linux 10 (buster)
Distribution: MainsailOS 0.7.1 (buster)

d3d3d, 2023-05-08:
Used to work now it does not.
Non blinking cursor at upper left corner on Raspberry 7" Touch Display.
klipper
v0.11.0-198-g33b18fd6
KlipperScreen
v0.3.2-27-g0b973ad
crowsnest
v3.0.7-3-g20ed6a8
moonraker
v0.8.0-40-gb21f177
sonar
v0.1.2-0-g7962a12
timelapse
v0.0.1-132-gf9198e0
mainsail
v2.5.1

TexasTomBF, 2023-05-09:
And same happening here to. The update broke it.

xArgentiusx, 2023-05-09:
The same issue here, but
KlipperScreen
v0.3.2-29-g49d910a 
last night klipperscreen worked, after switching the printer off, and this evening switching on klipperscreen does not work. only black screen with a little cursor in the top left edge

htgani, 2023-05-09:
I had the same issue, I fixed mine by moving file 99-fbturbo.conf from /usr/share/X11/xorg.conf.d/ to /home/pi.
ssh to pi and type "sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~"
and then reboot the pi

rosspirtle, 2023-05-09:
> I had the same issue, I fixed mine by moving file 99-fbturbo.conf from /usr/share/X11/xorg.conf.d/ to /home/pi.
> ssh to pi and type "sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~"
> and then reboot the pi

This worked!  Thank you!

jkaterenchuk, 2023-05-09:
Worked for me also.  TU @htgani

P3ps, 2023-05-10:
Thanks @htgani.
problem resolved!!

mintonette, 2023-05-10:
> I had the same issue, I fixed mine by moving file 99-fbturbo.conf from /usr/share/X11/xorg.conf.d/ to /home/pi. ssh to pi and type "sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~" and then reboot the pi

This worked for me.  Was printing at the time so got away with just restarting the Klipperscreen service instead of a full reboot.

derDoc81, 2023-05-11:
Worked for me as well. Thank you!

alfrix, 2023-05-14:
same as #928 in the sense that is an OS update that broke it, ks did not change anything that could cause this, i will leave this open for a while so people don't open the same issue again and again

LeeOtts, 2023-09-26:
> I had the same issue, I fixed mine by moving file 99-fbturbo.conf from /usr/share/X11/xorg.conf.d/ to /home/pi. ssh to pi and type "sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~" and then reboot the pi

Got me working again! THank you... well until I reboot so gonna keep this command around

HumbleDeer, 2023-10-13:
> I had the same issue, I fixed mine by moving file 99-fbturbo.conf from /usr/share/X11/xorg.conf.d/ to /home/pi. ssh to pi and type "sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~" and then reboot the pi

Does anyone have any information on what this regression is related to, and whether this fix is a short lived temporary thing or will require automatic or manual mitigation in the future (i.e. needs a doc page to help with it)?

alfrix, 2023-10-13:
> Does anyone have any information on what this regression is related to, and whether this fix is a short lived temporary thing or will require automatic or manual mitigation in the future (i.e. needs [...]

the fix should be https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/FBturbo/
which purges the packages (purge = remove package and config files)

it is seems that some installers like [LCD-show](https://github.com/goodtft/LCD-show) copy the file without checking if the driver was installed, the user may have to remove the file manually if the package was not installed or not completely removed, 

the fbturbo driver is not used anymore since it did not provide any benefits in performance in the last releases (at least not when i tested) the more standard fbdev is preferred

commit c14fd4d2 (2023-10-15): docs: add an extra step for fbturbo issues close #971
closed 2023-10-15 by alfrix