KlipperScreen/KlipperScreen #975: [Q] Turn Backlight off

Sineos, 2023-05-10:
I'm running KlipperScreen with [this display](https://www.aliexpress.com/item/1005005257187800.html) on an OrangePi 3 LTS with Armbian Bullseye. It is working beautifully except on thing: I have not found a way to also turn off the display's backlight when in screensaver mode.

Also:
`xset dpms force standby`
or 
`xset dpms force suspend`
will blank the display but always leaves backlight on.

Is there any way to turn it off?

In any case: Many thanks for KlipperScreen. Really a great addition to Klipper's ecosystem!

alfrix, 2023-05-12:
there are lots of displays that don't support turning off backlight via software

ks sets `xset -display :0 s blank` by default you could try to set `xset -display :0 s noblank`  some displays behave better accordingly to some users

i would try the full desktop OS (preferably on a spare sdcard/drive) if it doesn't work there then it probably doesn't have the feature

closed 2023-05-12 by Sineos

wiltwong, 2023-10-25:
It appears that klipper screen is setting `xset -display :0 dpms 0 {self.blanking_time} 0` which only sets that suspend time and not the off time. For my monitor, that setting only blanks the screen, I can manually turn off the backlight by running `xset dpms force off`. I didn't want to update the klipper screen code to accomadate just my monitor, so my work around is to run `xset -display :0 dpms 0 900 900` after klipper screen has started, that sets the suspend and off time to 15 mins. I do this with the klipper extention to run shell commands (https://github.com/dw-0/kiauh/blob/master/resources/gcode_shell_command.py) and the following macros/configuration:

```
[gcode_shell_command klipper_screen_dpms]
command: /usr/bin/xset -display :0 dpms 0 900 900
#  The linux shell command/script to be executed.  This parameter must be
#  provided
timeout: 2.
#  The timeout in seconds until the command is forcably terminated.  Default
#  is 2 seconds.
verbose: False
#  If enabled, the command's output will be forwarded to the terminal.  Its
#  recommended to set this to false for commands that my run in quick
#  succession.  Default is True.

[delayed_gcode _SET_KLIPPER_SCREEN_DPMS]
initial_duration: 5
gcode:
  RUN_SHELL_COMMAND CMD=klipper_screen_dpms
```

It's not perfect, if I change the klipper screnn settings, I need to maually run the command again or restart klipper, but it works for me.

alfrix, 2023-10-31:
idk if using the off-time will cause touchscreens to not wakeup, at least some tests on real hw should be done

at least these types (most common that i've seen) should be tested to make that change:
- [ ] hdmi with usb touch
- [ ] dsi type screens
- [x] waveshare 35bv2 DPI (GPIO) does work

wiltwong, 2023-10-31:
Agreed, that's why I didn't send a pull. FWIW, it works for me on my MIMO Vue HD-1080C-G USB touchscreen. In theory it should work since it's all controlled by the X server, but in practice it's going to depend on the hardware, both the video card/chip, it's driver and the attached display. 

Maybe I will send some time and add an additional configuration toggle for "DPMS screen off" that would at least give an option to use it? Instead of my workaround.