KlipperScreen/KlipperScreen #983: [FR] Troubleshooting guide, add disabling the TV out on Pi3

shiftingtech, 2023-05-22:
So I was recently setting up klipperscreen on a pi3 system, and once I got everything else working, had the interesting discovery that X was trying to output on my touch screen AND the tv out.

Video wise, this didn't really matter, but the actual TOUCH events for the touch screen got spread out across both, causing all the touch events  to be *grossly* mis-calibrated.

I'd suggest adding an entry to troubleshooting, suggesting adding
"enable_tvout=0" on pi3 and older.

alfrix, 2023-05-24:
i agree that the troubleshooting guide needs more work, i will give this priority since i've seen people having issues with OS updates

alfrix, 2023-05-28:
added to the docs eb24987b27c63526e41b6b2269ba49f6738bebe4

closed 2023-05-28 by alfrix