KlipperScreen/KlipperScreen #990: [BUG] KlipperScreen Touch not working on clean MainsailOS  1.2.0 (KIAUH install) - 7" Raspi touch screen w/ fix.

d3d3d, 2023-05-25:
KlipperScreen Touch not working on clean MainsailOS  1.2.0 (bullseye) using KIAUH install.

Steps to reproduce the behavior:
Install KlipperScreen

Expected behavior:
Touch screen should work.

Fix:
edit /boot/config.txt add two comment out (#) as shown.
#Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2

alfrix, 2023-05-25:
Not a bug, klipperscreen doesn't control the touchscreen, it's just an application, the proposed fix has already been reported, I will add it to the troubleshooting page but it will take a while since I'm rewriting and expanding.
related to  #928

HMidori, 2023-05-27:
> KlipperScreen Touch not working on clean MainsailOS 1.2.0 (bullseye) using KIAUH install.
>
> Steps to reproduce the behavior: Install KlipperScreen
>
> Expected behavior: Touch screen should work.
>
> Fix: [...]

I did a clean install of MainsailOS last month and added KlipperScreen.
While the display did show up correctly, the touchscreen did not work from the start.
"https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/" measures mentioned didn't solve anything, but with this solution the touchscreen now works. Thanks.
My KlipperScreen setup is Raspberry Pi 2B and  Waveshare 5inch DSI Display.

muzzlitebeer, 2023-05-27:
I'm having the same issue and the suggested fix didn't work for me.  I also followed all of the steps in the Troubleshooting guide.

alfrix, 2023-05-28:
added to the docs via eb24987b27c63526e41b6b2269ba49f6738bebe4

closed 2023-05-28 by alfrix