KlipperScreen/KlipperScreen #991: [FR] LED

Vasilyev81, 2023-05-25:
Hi, I'm Andrey.
From the beginning, I want you to know that I have no experience in contributing to an open source, so if I do something in an unusual/inappropriate way, do not hesitate to let me know about it, and please be patient.
In short, I have almost finished the development of the led panel, and just now I thought that (maybe) it is not implemented yet for some specific reasons.
The question is:  would you like to consider accepting a PR that implements led management functionality?
Below is a filled-out FR template from GitHub
**Is your feature request related to a problem? Please describe.**
It would be useful to be able to control the LED light from the KlipperScreen.

**Describe the solution you'd like**
I'm at the end of development and hope to finish it in the next few days. 

**Describe alternatives you've considered**
Currently, I have decided to go with scale for each color in rgb/rgbw mix; in the beginning, I considered using a color chooser widget, but I didn't find a way to customize it to RGBW color mix.

**Additional context**
Currently, as an environment, I have a Raspberry with a connected mouse and external screen and the second environment is  on Ender 3 V2 + Bigtreetech Manta E3EZ mainboard + Bigtreetech CB1 + sensor screen + neopixel led light.

Vasilyev81, 2023-05-26:
Here's how it looks right now, the LED panel is accessible from the actions menu[image]

[image]

[image]

[image]

alfrix, 2023-05-27:
hi, thank you for helping

>  What do you think about the UI layout? Is it okay?

a color-picker would be preferred, there are some ways to convert to rgbw search online

> Is it necessary to implement management of more than one RGB/RGBW led device?

Yes, but starting with one is ok.

> How common is it to have more than one RGB/RGBW led device?

idk we don't have statistics that i know

Vasilyev81, 2023-05-27:
Okay, I get it. Will look for rgbw color picker.
Thanks!

Vasilyev81, 2023-06-01:
Hi, I'm kind of stuck
When I'm adding ColorChooserWidget in this way:
`self.color_chooser = Gtk.ColorChooserWidget()
  self.color_chooser.set_vexpand(True)
  self.color_chooser.set_hexpand(True)
  self.color_chooser.set_valign(Gtk.Align.FILL)
  self.color_chooser.set_halign(Gtk.Align.FILL)
  self.content.pack_start(self.color_chooser, True, True, 0)`

it is not fill all available space in the self.content
What I'm doing wrong? Any advice/help appreciated

alfrix, 2023-06-02:
i have come across that issue, try placing it inside a grid it usually solves it

jay1st, 2023-06-12:
the FLSUN pad has 2 LED buttons, can we have that added or a HOWTO ?

Vasilyev81, 2023-06-12:
Hi! Did you mean Flsun Speeder Pad

jay1st, 2023-06-12:
> Hi! Did you mean Flsun Speeder Pad

Oh yes exactly

floodo1, 2023-07-13:
> Okay, I get it. Will look for rgbw color picker. Thanks!

Fluidd uses this color picker. I find the circle (hue/saturation) and slider (brightness) works well. Would love to see something similar in Klipperscreen!

[image]

Vasilyev81, 2023-09-18:
> So I revise my statement: "Would love to see _any led interface_ in Klipperscreen!"

Soon)

commit 87d09036 (2023-09-24): Led light control, close #991   (#1106)

* feature: add leds panel

* Various refactors and improvements

* lint fixes

* led: use more columns for the selector

* leds: name in the title

* mainsail presets

* change sliders to horizontal to avoid visual bug, simplify defaults

* led: add a color preview and simplify methods to fix some issues

* simplify presets move all the logic to the panel

* led: change presets from names to previews

* preview label shows color, add vertical mode

* led: colored previews as rectangles instead of circles

---------
closed 2023-09-24 by alfrix