KlipperScreen/KlipperScreen #996: [BUG] Memory leak from print dialog

simpat1zq, 2023-05-29:
I'm still having the issue from the closed bug:
https://github.com/jordanruthe/KlipperScreen/issues/932

Everytime I open the print dialog after each print, it adds about 1 to 2% of memory usage. So klipperscreen causes the pi to crash, and fail a print every couple of weeks.

alfrix, 2023-06-13:
it should be fixed in v0.3.2, it wasn't the print dialog that leaked it was the transition to and from job_status (printing screen)

closed 2023-06-13 by alfrix

simpat1zq, 2023-06-13:
I've got good news to report. I just opened the print files dialog. I have a TON of gcode files, and it used to take like 4 seconds to open, now it opens in about 1 to 2. Then I switched printers, and opened the print files dialog, and repeated that about 20 times. Memory usage stayed perfectly steady after opening them the first time. And after switching back, the print files dialog opens pretty much instantly. 

Previously, the dialog would take longer and longer to open while the memory fills up. And this same test would have filled up my memory after switching about 40 times. I still have to make sure that memory doesn't inch up after prints and stuff, but I'm confident that you found a fix. This update is HUGE for me. No more waiting for the print dialog to load. Thank you so much.