KlipperScreen/KlipperScreen #999: [BUG] Klipperscreen loses sync with klipper using "bound_services: klipper" in moonraker power device config

jwarner3412, 2023-05-30:
After adding "bound_services: klipper" to [power printer] in moonraker.conf, klipperscreen hangs or desyncs with klipper. 

For instance, after my print_start macro, it runs an adaptive mesh (which throws the save_config flag). Print finishes, printer times out and shuts off. "Save_config?" screen still active on KS. If i start another print, printer turns on, print begins, "Save_config?" screen still up. If i cancel the prompt it brings me to the idle dashboard screen. Temps on the idle screen don't align with the interface either. If i choose save, it reboots the printer and interrupts the print that has just started. I've even had KS be on the idle dashboard screen, send a print from the slicer, print starts, KS still on the idle dash but flashing the gcode commands. Even if i interact with the "Save_config?" before the printer times out it will still be bugged when it comes back online. I have to restart the klipperscreen service with every print if i want to use the screen.

Am i missing a setting somewhere? If i bound klipperscreen to the power device in moonraker it should work to print but i wouldn't be able to turn the printer on through klipperscreen (because the service wouldn't be running) I'm assuming. I found one other person online with this issue and he made a gcode macro to restart klipperscreen. Worst case i can add that to my print_start macros but it doesn't seem to be desired behavior.

My config:
https://github.com/jwarner3412/en3p_klipper_cfg

Steps to reproduce the behavior:
1. add "bound_services: klipper" to [power printer] in moonraker.conf (along with the appropriate settings for idle power off and auto power on if jobs are added to que)
2. let printer idle timeout and power off
3. send print to printer via slicer using [job_queue] to power on printer automatically.
4. try to interact with print screen

**Expected behavior**
klipperscreen to function as normal

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11605589/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/11605590/klippy.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/11605591/moonraker.log)

franz134, 2023-06-07:
same issue here... I've pretty much the same config because I like the mainsail printer powered off instead of the red error message.
I have found out that a KS restart works, so (if the save config pops up cancel) and from the idle screen>settings>system>restart KS. At least I'm my case I get back to the print screen.

jwarner3412, 2023-06-07:
I've resorted to using delayed gcode to restart ks when klipper starts. You need the gcode shell command addon in kiauh installer (advanced section). I'll push my recent local changes to GitHub, I haven't had an issue since, just seems very hacky lol.

closed 2023-06-13 by alfrix