KlipperScreen/KlipperScreen #1001: [FR] up/down arrows

guestisp, 2023-06-01:
Please add up and down arrows along the scrollbars, sometimes, when you have to scroll a little amount, would be easier to press the up or down botton twice than start dragging a scroll bar (that on some screen is small)

Vasilyev81, 2023-06-02:
HI! On my 3.5-inch display scrollbar is also tiny, I feel you. First of all, I'm not a maintainer of this repo, but in case you want to try if it will be easier for you to use the buttons, you can try adding this code to KlipperScreen/styles/base.css
Values 2em is close to the current size of the scrollbar, optionally, you can change all of them to 3em, and it'll become wider(taller for horizontal scrollbar). If this repo's owner decides these are reasonable changes, I can put it in a PR.
`
scrollbar.vertical slider,
scrollbar.slider.vertical {
    min-width: 2em;
}

scrollbar.horizontal slider,
scrollbar.slider.horizontal {
    min-height: 2em;
}

* {
    -GtkScrollbar-has-backward-stepper: 1;
    -GtkScrollbar-has-forward-stepper: 1;
}

scrollbar button {
    min-width: 2em;
    min-height: 2em;
}

scrollbar.vertical button.down {
    -gtk-icon-source: -gtk-icontheme("pan-down-symbolic");
}

scrollbar.vertical button.up {
    -gtk-icon-source: -gtk-icontheme("pan-up-symbolic");
}

scrollbar.horizontal button.down {
    -gtk-icon-source: -gtk-icontheme("pan-end-symbolic");
}

scrollbar.horizontal button.up {
    -gtk-icon-source: -gtk-icontheme("pan-start-symbolic");
}
`

guestisp, 2023-06-04:
Absolutely awesome. Please add it as default.

(maybe could be adjusted to increase the scroll amount, actually it scroll very very slow)

guestisp, 2023-08-05:
@alfrix could you please add something similiar ? I can't upgrade KlipperScreen due to this customization (that works, and is very useful if you have to scroll a lot o files)

closed 2023-08-18 by alfrix