KlipperScreen/KlipperScreen #1002: KS stuck on consolle no screen found

MIRAEngines, 2023-06-01:
Hello,

Im having trouble with KS on BTT PITFT 50 and RP 3a+

Installation was ok but im stuck on the console.  The display itself is working as it should.

I have tried every possible fix I could find on the web, but sadly there my abilities end.

? KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-06-01 15:46:45 BST; 10s ago
   Main PID: 2269 (KlipperScreen-s)
      Tasks: 3 (limit: 255)
        CPU: 90ms
     CGroup: /system.slice/KlipperScreen.service
             tq2269 /bin/bash /home/pi/KlipperScreen/scripts/KlipperScreen-start.sh
             mq2272 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py

Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE)
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: Fatal server error:
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE) no screens found(EE)
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE)
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: Please consult the The X.Org Foundation support
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]:          at http://wiki.x.org
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]:  for help.
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE)
Jun 01 15:46:45 mainsailos KlipperScreen-start.sh[2273]: (EE) Server terminated with error (1). Closing log file.

[Xorg.0.log](https://github.com/jordanruthe/KlipperScreen/files/11626762/Xorg.0.log)

Im sure that I have done something wrong, I have reinstalled the KS via the kiauh and added the moonraker config info per the installation instructions. 

[image]

Thank you for a possible solution.
Milan

alfrix, 2023-06-09:
[  3600.135] Parse error on line 1 of section InputClass in file /usr/share/X11/xorg.conf.d/99-fturbo.conf
	"/dev/fb0" is not a valid keyword in this section.
[  3600.135] (EE) Problem parsing the config file
[  3600.135] (EE) Error parsing the config file

it seems that you may have messed around with 99-fturbo.conf please paste it here

alfrix, 2023-06-13:
closing because of no response, preferably only use github issues for bugs or feature requests not for install support

SilverBulletCoder, 2024-09-06:
I know this is closed, but i spent way to many hours to stumble apon this same issue, and i just fixed it.
I tried to go the VNC Route, but ended up not liking it.  Tiger screwed my install up so bad.
the 99-fturbo.conf file was wiped out.  I removed "/dev/fb0" then pasted the following:

Section "Device"
    Identifier "FTDI"
    Driver "fbdev"
    Option "fbdev" "/dev/fb0"
EndSection

This resolved my issue and KlipperScreen is back to working on my touch screen.