KlipperScreen/KlipperScreen #1006: [BUG] Screen changes its size

Vasilyev81, 2023-06-08:
**Describe the bug**
On opening the console screen changes its dimensions

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Actions'; then go to "Console"
2. The screen stretches vertically, and its lower part extends beyond the screen.

**Expected behavior**
The screen saves its size. 

**Screenshots**
https://github.com/jordanruthe/KlipperScreen/assets/26001746/ce7dbc48-670c-453e-aff2-99c6c78c00b4

**Logfile**
I've tried to identify the bug and added a logging command at screen.py at the update_size function at line 972
logging.error(f"Event 'configure_event' fired, size: {self.get_size()}")
The attached log file is exactly related to the video I've attached in  **Screenshots**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/11685789/KlipperScreen.log)

alfrix, 2023-06-09:
confirmed will be fixed in the next release

closed 2023-06-13 by alfrix

Vasilyev81, 2023-06-13:
Hi!
How to make a PR? I mean when I try to push git says that I don't have enough permissions or smth like that. I want to close 991 and 1001.

alfrix, 2023-06-13:
you can then go to your local folder and fetch > rebase > push