KlipperScreen/KlipperScreen #1007: [BUG] Asking for login password

clsferguson, 2023-06-09:
**Describe the bug**
When booting up, the screen shows the typical linux boot up messages, then it gets to the login prompt and stops there.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)
[klipperscreen.log](https://github.com/jordanruthe/KlipperScreen/files/11697414/klipperscreen.log)

the main board is a manta m4p, I am using a CB1 with BTT's CB1 debian image. I updated all the firmwares on everything it is packaged with. It was working after updates, but seemingly not after power cycle. the default login and pass is different from a pi for the BTT's image. biqu/biqu.

I have checked [this](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/VC_ERROR/l) stuff. No it is all there.

alfrix, 2023-06-09:
from the log klipperscreen is working correctly, your os is problably outputting on the wrong framebuffer
it's usually people not following the correct install for those gpio screens, i already described the issue here
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Showing_console/#lcd-show
and the manual method on the bottom
closing as invalid since it's not a klipperscreen bug, it's a OS config issue

closed 2023-06-09 by alfrix

alfrix, 2023-06-09:
i pointed to LCD-Show because the link in reddit you mentioned is having that problem.

now with the xorg logs  i can see that it's a driver issue
it seems that there is an issue with the version of mesa you are using, you need to upgrade or downgrade it.

clsferguson, 2023-06-09:
Ok, so I think I resolved it, I kind of got lucky, I was about to reflash the OS. I installed kiauh. Then I removed KlipperScreen, reinstalled it. I did a host reboot, the issue was still present. So, I shut it down. Then I remembered to copy my printer.cfg. When I booted it back up again, the screen is now working.