KlipperScreen/KlipperScreen #1012: [BUG] Wrong print file dialog after switching printers, it shows the dialog from the first printer opened.

simpat1zq, 2023-06-13:
From my issue I opened for the memory leak, I think that the fix for that might be causing this bug, but that's just a guess. Here are the steps to reproduce:

Start Klipperscreen pointing to Printer A. 
Tap Print
Tap switch printers, and switch to Printer B. 
Tap Print. 

The print dialog should be for printer B, but it is not. It just loads the print dialog for printer A. Not sure if the same thing is happening for other dialogs or not.

alfrix, 2023-06-13:
since i don't have multiple printers i don't test that, but that sounds very bad, i will give it priority

simpat1zq, 2023-06-13:
This is supposed let you have a dummy printer. I've never used it, but that might allow you to test some stuff.

commit 92ed72e4 (2023-06-16): fix panel removal not working correctly

previous fix addresed the leaking memory, but created more issues:
fixes #1012
fixes #1015
fixes #1017
closed 2023-06-16 by alfrix