KlipperScreen/KlipperScreen #1015: [BUG] The language switching function is damaged.

bigtreetech, 2023-06-16:
**Describe the bug**
* change the language to Chinese in the settings interface. Only some entries in the interface have been changed to Chinese, while others remain in English. And after reboot, all entries are restored to the default English. (Tested other languages and had the same problem)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Congifuration->Settings->Language`
2. Choose a language other than English
3. It is easy to find that there is no language switching for many entries
4. Restore all entries to default English after reboot

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

bigtreetech, 2023-06-19:
@alfrix Hi. After switching languages, all entries can be switched to the set language normally, This problem has been fixed. 

However, the issue of automatically changing back to the default English after reboot still exists.

mehmetsutas, 2023-07-02:
Confirmed. Language setting is not saved. Even after saving the settings reboot automatically enables English.

alfrix, 2023-07-02:
sorry i did not notice the response that it wasn't fully fixed, i will look into it

commit 92ed72e4 (2023-06-16): fix panel removal not working correctly

previous fix addresed the leaking memory, but created more issues:
fixes #1012
fixes #1015
fixes #1017
commit 0f47d452 (2023-07-02): fix language not saving fixes #1015
closed 2023-07-02 by alfrix