KlipperScreen/KlipperScreen #1032: bug/feature request.

danorder, 2023-07-03:
Problem / solution 
z-calibrate:  probe_calibrate ,delta_calibrate/manual   do not reset set_gcode offset 

If using macros from klipper its possible to have semi different behavior. e.g autosaving zoffset (flsun printers)
In this case macros  would reset gcode_offset aka babystep However klipperscreen does not leaving the previous offset in tact. 
Steps to reproduce would be having a previous gcode_offset  then running "klipperscreens" calibration buttons leading to a nozzle to high or bed scrape.  So in short the request is to reset the offset on all commands to avoid this scenario.   Thanks in advance.

alfrix, 2023-07-08:
if i understand correctly the issue comes with gcode z offset not being reset back to 0 after a probe_calibrate
this is usually not a problem since most users will save configuration and that resets the offset back to 0

this is a thing that klipper should do and not klipperscreen, the place to ask for it is https://klipper.discourse.group/c/general-discussion (also see https://www.klipper3d.org/Contact.html#i-found-a-bug-in-the-klipper-software)

alfrix, 2023-07-08:
as an inmediate workaround you can use macros like:
```
[gcode_macro PROBE_CALIBRATE]
rename_existing: PROBE_CALIBRATE_1
gcode:
  SET_GCODE_OFFSET Z=0
  PROBE_CALIBRATE_1 { rawparams }
```

closed 2023-07-08 by alfrix

danorder, 2023-07-10:
The problem would be most models off flsun delta printer use a version of
mentals' zoffset macro. which saves the last used babystep value to
variables file. so if i had a babystep of plus 0.2  this wont change until
its manually reset by hand, in the copy i support a calibrate macro which
already handles this situation similar to your suggestion. However it does
not use a rename. since in the past this lead to crashes.running from the
screen. So historically its seemed safer to keep the commands "vanilla" to
avoid unintended conflicts with projects / docs. due to the number of
printer models using this im also unable to change the default
functionality the user base already knows. Since a delta needs a clean
slate every calibration cycle regardless i respectfully. disagree since in
my opinion it should resetting to "defaults" and, i cant see a downside to
having a failsafe perhaps similar to the slight zhop workaround.