KlipperScreen/KlipperScreen #1035: [BUG]

bigtreetech, 2023-07-06:
**Describe the bug**
https://klipperscreen.readthedocs.io/en/latest/Hardware/#screen-rotation
According to the description here, setting the screen rotation vertical screen does not work.
My screen resolution is 1024 * 600.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

alfrix, 2023-07-08:
rotation is handled by the os, this is probably a problem with the display drivers, but at least you need to provide the logs to see if something is up with klipperscreen itself or if its something in X

bigtreetech, 2023-07-11:
Got it. Thanks.
It seems that it is indeed a OS issue and is being further identified. Once I find the problem, I will synchronize the detailed information here. 

Now close this issue to avoid misleading

closed 2023-07-11 by bigtreetech