KlipperScreen/KlipperScreen #1042: [FR] Custom GCode before/after manually extruding

garyd9, 2023-07-09:
**Is your feature request related to a problem? Please describe.**
At times, when manually loading filament via the extrude/retract functions (not the filament_load/unload macros), my filament runout sensor triggers.  

**Describe the solution you'd like**
If klipperscreen allowed providing custom gcode that was run before and after manual extrude/retracts, I could disable the filament sensor before extruding and re-enable it afterwards. 

**Additional context**
An alternative would be for klipperscreen to support turning on/off a filament sensor, but it would still be easier to have custom gcode before/after manual extrudes.  For example, before manually extruding:

SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0

and after manually extruding:

SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1

alfrix, 2023-08-03:
turning off the filament sensor is supported why do you think there is a giant slider

[image]

garyd9, 2023-08-04:
Perhaps the filament sensor status isn't being properly read from klipper.  Quite often, when I go to the "extrude" page, the slider is on the left and the banner is dark, then I'll extrude without running through the filament sensor, and the printer detects a runout.  

If I use that slider to turn on the sensor (which moves the slider to the right and changes the color of the banner) and then turn it back off (which moves the slider to the left and makes the banner dark again), and then extrude, I don't have the same problem.

closed 2023-10-03 by alfrix