KlipperScreen/KlipperScreen #1043: The MUVE error

FastivProService, 2023-07-09:
Good day.
There is a problem. Maybe someone will help.  menu does not work. Thanks

[image]
[image]

alfrix, 2023-07-13:
there is not enough information, i want to see the full log file to see if there is more clues in there.
there maybe an issue with reading the klipper configuration, i want to see it too
attach the log and klipper config as zip

FastivProService, 2023-07-14:
[KlipperScreen.conf_1689358000810.zip](https://github.com/KlipperScreen/KlipperScreen/files/12052857/KlipperScreen.conf_1689358000810.zip)

FastivProService, 2023-07-14:
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/12052859/KlipperScreen.log)

alfrix, 2023-07-15:
the only thing that can cause this according to my recent testing, is failure to get max_velocity from the printer, that is why i asked for the **_klipper_** config

FastivProService, 2023-07-15:
[1689437340991.zip](https://github.com/KlipperScreen/KlipperScreen/files/12062077/1689437340991.zip)
thanks for helping me

alfrix, 2023-07-15:
as i suspected the issue is 
```max_z_velocity: 1```
min and max cannot be equal, i will make a failsafe for this case

FastivProService, 2023-07-15:
All right. Changed to max_z_velocity 2 and the error disappeared. Thank you.

commit 432305be (2023-07-30): move: add a failsafe for velocity fixes #1043
closed 2023-07-30 by alfrix