KlipperScreen/KlipperScreen #1059: [BUG]

tekgamerpt, 2023-07-25:
**Describe the bug**
Screen resolution not filling BTT HDMI 5 inch screen after 180º rotation

**To Reproduce**
Steps to reproduce the behavior:
1.Rotate 180º screen
2. Reboot Pi
3. Resolution not filling all screen, missing a bit on left side. Touchscreen running good on full screen size

alfrix, 2023-07-25:
klipperscreen is just an application you need to configure the resolution in the operating system

closed 2023-07-25 by alfrix

tekgamerpt, 2023-07-25:
Yeah but, without screen rotation everything works fine. Do i need to change resolution after screen rotation 180º?

alfrix, 2023-07-25:
in the klipperscreen log file you should have the detected resolution, if you want me to check, post it here

in the docs i have written some guides about rotation but every os and hw tends to behave differently:

https://klipperscreen.readthedocs.io/en/latest/Hardware/#screen-rotation

tekgamerpt, 2023-07-25:
That was the guide i used