KlipperScreen/KlipperScreen #1060: [FR] Spoolman support

blondak, 2023-07-26:
I started creating spoolman support at https://github.com/blondak/KlipperScreen

Any suggestions and Ideas welcome

Mascherino, 2023-07-26:
I'm working on spoolman support as well. Is there a reason why you're creating classes and saving everything as attribute when the moonraker server already provides the data in json format?

blondak, 2023-08-03:
I'm used to, I like strict types and even in classes one can enrich with getters or setters, so no particulat reason.

blondak, 2023-08-03:
For example you can create a property (icon) https://github.com/blondak/KlipperScreen/blob/master/panels/spoolman.py#L111 with getter get_icon and later in code use https://github.com/blondak/KlipperScreen/blob/master/panels/spoolman.py#L217

Mascherino, 2023-08-04:
I did not mean to critizise you in any way. I was just curious as I don't think you need to save all the attributes for every spool. But it also depends on how far you want to take the integration for spoolman. I personally think a simple way to choose the currently active spool from a list would be enough.

blondak, 2023-08-06:
how it looks right now ;)
[image]

blondak, 2023-08-06:
Added filtering by material
[image]

commit af1de12f (2023-08-19): Spoolman support close #1060

Refactor post_requrest
Switching to TreeView
More spool info
redesign toggle buttons
Allow spoolman while printing
Styling checkbox as switch
render Toggle switch-like
Fixed wrong timezones
Filtering spools
Removed translations of Spoolman
Spool image can be overided in theme
Added switch-on/off symbols to toggle
rename iter to i to avoid shadowing built-in function
remove redundant parenthesis
convert spoolcompare to static
merge check and treeview shared props
reduce spacing of buttons very useful for vertical/portrait
fix issue with remainings
add spoolman shor [... cut]
closed 2023-08-19 by alfrix