KlipperScreen/KlipperScreen #1061: [FR] Run on VNC + VNC client

dottedmag, 2023-07-26:
**Is your feature request related to a problem? Please describe.**

I'd like to use old iPad as a control panel for Klipper. However its Safari (and thus all the other browsers) is so old it can't render Mainsail or Fluidd.

**Describe the solution you'd like**

- Run KlipperScreen on X VNC server
- Run VNC client on iPad.

I've made it work, the procedure below needs some polishing & inclusion into documentation.

**Describe alternatives you've considered**

- Weston+VNC. This is just committed to Weston, not available in stable linux distros.
- Weston+RDP. Turned out KlipperScreen has a number of hardcoded X11-specific calls.
- X11+RDP. Xrdp turned out to be very hard to set up.

**Additional context**

I've done the following to make it work:

On a host:
- Install KlipperScreen via KIAUH on Debian 12.
- Install `tigervnc-standalone-server` package.
- Set the following configuration:
```
[main]
screen_blanking = off # Should be set on a client instead
use_dpms = False # not supported by Xtigervnc anyway
```
- Create `launch_KlipperScreen.sh`:
```
#!/usr/bin/env bash

# Use display 10 to avoid clashing with local X server, if anyy
Xtigervnc -rfbport 5900 -noreset -AlwaysShared -SecurityTypes none :10&
DISPLAY=:10 $KS_XCLIENT&
wait
```
- Start `KlipperScreen.service`

On iPad:
- Go to `Settings`
  - Go to `General`
    - Set "Use side switch to" to "Lock Rotation". This will prevent unwanted rotation of UI.
    - Go to `Restrictions`
      - Set passcode
      - Enable restrictions. This will be needed to avoid accidentally switching between apps.
- Install `RealVNC Viewer: Remote Desktop` app
- Open VNC viewer
- Triple-click "Home" button
  - Guided access pops up
  - Press "Start"
  - Now iPad is locked to VNC viewer until "Guided access" mode is disabled by triple-clicking "Home" button and entering the restrictions password.
- Press "+" button at the top right
- Enter IP address of your print host.
- Press "Save"
- Select "Interaction", select "Touch panel", go back
- Press "Done"
- Double-click on an icon with IP address you have just added.
- VNC client will complain about unencrypted connection. Disable the warning and say "Connect"
- Use or skip tutorial (if you've seen it already)
- Press the "Pin" icon to hide the panel.
- Set the side switch to lock the UI rotation
- Enjoy!

dottedmag, 2023-07-26:
iPad steps are described using iPad Mini (2012) and iOS 9.3.6. They might differ for other hardware (e.g. if no side switch is present, or if Retina display is in use).

alfrix, 2023-07-29:
thank you, the guide needs to be written in markdown and generalized a bit since vnc can be used by many devices, the ipad can be left as example

i maybe do this today since it's quick.

> Turned out KlipperScreen has a number of hardcoded X11-specific calls.

KS works on wayland, it has 2 minor issues:
1. can't set dpms and will use internal timers to turn the screen black (no xset on wayland)
2. it will show cursor on settings dropdowns, xsetroot was needed on x11 to hide cursor on dropdowns

to use wayland the installer needs to be modified to offer weston in kiosk mode
xset can be replaced by `weston -i`

i haven't made any attempt to make the installer mods (which will be optional), i am just waiting for raspberry-os bookworm which will supposedly have wayland support, and i'm not sure it will be really better

closed 2023-07-30 by alfrix