KlipperScreen/KlipperScreen #1062: [FR] Specify default printer when multiple are defined

rdbahm, 2023-07-29:
**Is your feature request related to a problem? Please describe.**
I own two Klipper printers, but only one runs a KlipperScreen. While eventually I'll give in and buy a touchscreen for the other printer, configuring both printers on the KlipperScreen-equipped machine suits my use-case well, since it gives me all the capabilities of KS for both printers with no additional cost. However, 90% of the time I'm interested in interacting with the connected machine, not the remote machine.

**Describe the solution you'd like**
I would like to be able to specify a "default" parameter on a printer configured for KlipperScreen to have that printer automatically selected on KlipperScreen start.

**Describe alternatives you've considered**
Buying another screen is far too simple and elegant of a solution for me, because I simply _must_ overcomplicate things. With multiple printers in one area, a single KlipperScreen instance really is enough for my (and hopefully others') use-case.

**Additional context**
None

alfrix, 2023-07-29:
already possible 
https://klipperscreen.readthedocs.io/en/latest/Configuration/#main-options

closed 2023-07-29 by alfrix

rdbahm, 2023-07-29:
Thanks, I swear I read this page a half dozen times looking for it. You're a hero.