KlipperScreen/KlipperScreen #1074: [BUG] temperature graph not show

CIS94500, 2023-08-08:
### What happened?

hello, the temperature graph is not displayed the first time klipperscreen is started, you have to restart klipperscreen for it to be displayed.

### What did you expect to happen instead?

that the graph is displayed at the first start

### How to reproduce this bug?

start the printer and the pi

### Additional information:

[image]

### Log output

```shell
2023-08-08 00:12:05,940 [config.py:create_translations()] - Selected lang: fr OS lang: fr_FR
2023-08-08 00:12:05,941 [config.py:install_language()] - Using lang fr
2023-08-08 00:12:05,982 [screen.py:__init__()] - Screen resolution: 1024x600
2023-08-08 00:12:05,986 [KlippyGtk.py:__init__()] - Font size: 20.7 (medium)
2023-08-08 00:12:06,904 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 900
2023-08-08 00:12:06,924 [screen.py:set_screenblanking_timeout()] - Using DPMS
2023-08-08 00:12:06,968 [screen.py:initial_connection()] - Default printer: FLSUN SR
2023-08-08 00:12:06,969 [screen.py:connect_printer()] - Connecting to printer: FLSUN SR
2023-08-08 00:12:06,970 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-08-08 00:12:07,085 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-08-08 00:12:07,104 [KlippyWebsocket.py:connect()] - Attempting to connect
2023-08-08 00:12:07,139 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-08 00:12:07,162 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2023-08-08 00:12:07,165 [KlippyWebsocket.py:connect()] - Starting websocket thread
2023-08-08 00:12:07,183 [_logging.py:info()] - Websocket connected
2023-08-08 00:12:07,185 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2023-08-08 00:12:07,215 [screen.py:update_size()] - Size changed: 1024x600
2023-08-08 00:12:07,281 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-08 00:12:07,392 [KlippyRest.py:send_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/machine/device_power/devices
2023-08-08 00:12:07,420 [KlippyRest.py:send_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/printer/info
2023-08-08 00:11:53,789 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-08 00:11:53,829 [KlippyRest.py:send_request()] - 404 Client Error>>Not Found for url>>http://127.0.0.1:7125/machine/device_power/devices
2023-08-08 00:11:53,871 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2023-08-08 00:11:53,912 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2023-08-08 00:11:53,920 [printer.py:reinit()] - Klipper version: v0.11.0-257-ged66982b
2023-08-08 00:11:53,920 [printer.py:reinit()] - # Extruders: 1
2023-08-08 00:11:53,921 [printer.py:reinit()] - # Temperature devices: 3
2023-08-08 00:11:53,922 [printer.py:reinit()] - # Fans: 2
2023-08-08 00:11:53,923 [printer.py:reinit()] - # Output pins: 0
2023-08-08 00:11:53,924 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2023-08-08 00:11:54,211 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks&motion_report&firmware_retraction&exclude_object&manual_probe&extruder&heater_bed&temperature_sensor Orange_pi&temperature_sensor Super_Racer&fan&heater_fan heatsink_fan&filament_switch_sensor filament_sensor
2023-08-08 00:11:54,292 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-08-08 00:11:54,294 [printer.py:init_temp_store()] - Temp store: []
2023-08-08 00:11:54,313 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-08-08 00:11:54,315 [screen.py:init_tempstore()] - Temperature store size: 1200
2023-08-08 00:11:54,316 [files.py:initialize()] - Gcodes path: /home/pi/printer_flsun/gcodes
2023-08-08 00:11:54,317 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2023-08-08 00:11:54,319 [screen.py:init_printer()] - Printer initialized
2023-08-08 00:11:54,321 [printer.py:change_state()] - Changing state from 'disconnected' to 'startup'
2023-08-08 00:11:54,323 [printer.py:change_state()] - Adding callback for state: startup
2023-08-08 00:12:08,951 [printer.py:change_state()] - Changing state from 'startup' to 'ready'
2023-08-08 00:12:08,952 [printer.py:change_state()] - Adding callback for state: ready
2023-08-08 00:12:08,957 [screen.py:_load_panel()] - Loading panel: main_menu
2023-08-08 00:12:09,087 [main_menu.py:__init__()] - ### Making MainMenu
2023-08-08 00:12:09,099 [main_menu.py:add_device()] - Adding device: extruder
2023-08-08 00:12:09,112 [main_menu.py:add_device()] - Adding device: heater_bed
2023-08-08 00:12:09,126 [main_menu.py:add_device()] - Adding device: temperature_sensor Orange_pi
2023-08-08 00:12:09,138 [main_menu.py:add_device()] - Adding device: temperature_sensor Super_Racer
2023-08-08 00:12:09,182 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-08 00:12:09,185 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-08 00:12:09,212 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-08 00:19:19,642 [screen.py:_go_to_submenu()] - #### Go to submenu config
2023-08-08 00:19:19,643 [screen.py:_go_to_submenu()] - #### Menu __main
2023-08-08 00:19:19,654 [warnings.py:_showwarnmsg()] - /home/pi/KlipperScreen/panels/main_menu.py:93: Warning: Source ID 72 was not found when attempting to remove it
  GLib.source_remove(self.graph_retry_timeout)

2023-08-08 00:19:19,655 [screen.py:_load_panel()] - Loading panel: menu
2023-08-08 00:19:19,889 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > config
2023-08-08 00:19:19,908 [menu.py:arrangeMenuItems()] - X > config bedmesh
2023-08-08 00:19:19,923 [menu.py:arrangeMenuItems()] - X > config zcalibrate
2023-08-08 00:19:19,936 [menu.py:arrangeMenuItems()] - X > config bedlevel
2023-08-08 00:19:22,423 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-08 00:19:22,462 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-08 00:19:22,468 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-08 00:19:26,211 [screen.py:_load_panel()] - Loading panel: print
2023-08-08 00:19:26,305 [print.py:__init__()] - 24h time is True
2023-08-08 00:19:26,337 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > print
2023-08-08 00:19:26,339 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2023-08-08 00:19:29,046 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-08 00:19:29,075 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-08 00:19:29,079 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-08 00:19:33,096 [screen.py:_load_panel()] - Loading panel: temperature
2023-08-08 00:19:33,190 [temperature.py:add_device()] - Adding device: extruder
2023-08-08 00:19:33,204 [temperature.py:add_device()] - Adding device: heater_bed
2023-08-08 00:19:33,225 [temperature.py:add_device()] - Adding device: temperature_sensor Orange_pi
2023-08-08 00:19:33,237 [temperature.py:add_device()] - Adding device: temperature_sensor Super_Racer
2023-08-08 00:19:33,249 [temperature.py:select_heater()] - Selecting extruder
2023-08-08 00:19:33,252 [temperature.py:select_heater()] - Selecting heater_bed
2023-08-08 00:19:33,290 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > temperature
2023-08-08 00:19:35,128 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-08 00:19:35,157 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-08 00:19:35,164 [main_menu.py:update_graph_visibility()] - Could not create graph <Gtk.Box object at 0xa4429988 (GtkBox at 0x18894a8)> []
2023-08-08 00:19:35,166 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-08 00:19:53,365 [printer.py:change_state()] - Changing state from 'ready' to 'shutdown'
2023-08-08 00:19:53,368 [printer.py:change_state()] - Adding callback for state: shutdown
2023-08-08 00:19:53,538 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
```

alfrix, 2023-08-08:
your log is missing the start without it i do not know what version are you running

CIS94500, 2023-08-09:
This is the latest version, I did the update 3 days ago

CIS94500, 2023-08-09:
KlipperScreen version: v0.3.4-13-g0a3dfbe-dirty

alfrix, 2023-08-13:
i refactored the code that controls the graph visibility, it will be merged in the next version

CIS94500, 2023-08-15:
Hello and thank you for your work, I applied the 7d4a450 patch but the problem persists.

to reproduce the problem:

- turn off the printer and the pi
- start the pi and wait for the display of KlipperScreen
- turn on the printer
- click on restart klipper

```
--------------------KlipperScreen Log Start--------------------
Git Version: ?
2023-08-15 22:16:22,884 [screen.py:main()] - KlipperScreen version: v0.3.4-13-g0a3dfbe-dirty
2023-08-15 22:16:23,237 [config.py:get_config_file_location()] - Passed config (-c): /home/pi/KlipperScreen.conf
2023-08-15 22:16:23,239 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen.conf
2023-08-15 22:16:23,283 [config.py:log_config()] -  ===== Config File =====
[main]
default_printer = FLSUN SR
language = fr

[preheat PLA]
bed = 50
extruder = 200

[preheat ABS]
bed = 90
extruder = 245

[preheat PETG]
bed = 80
extruder = 240

[preheat FLEX]
bed = 0
extruder = 210

[printer FLSUN SR]
moonraker_host = 127.0.0.1
moonraker_port = 7125
move_speed_xy = 100
move_speed_z = 100
extrude_distances = 10, 20, 40, 80
extrude_speeds = 2, 4, 6, 8

[printer LK4 PRO]
moonraker_host = 127.0.0.1
moonraker_port = 7126
extrude_distances = 10, 20, 40, 80
extrude_speeds = 2, 4, 6, 8
=======================
2023-08-15 22:16:23,301 [config.py:__init__()] - ====== Saved Def ======

[main]
language = fr
screen_blanking = 900
side_macro_shortcut = False
show_heater_power = True
print_sort_dir = date_desc

[displayed_macros FLSUN SR]
cancel_print = False
load_filament = False
m204 = False
m600 = False
select_pa = False
set_gcode_offset = False
pause = False
resume = False
unload_filament = False
m205 = False
[... 83 lines cut ...]
2023-08-15 22:16:20,126 [main_menu.py:add_device()] - Adding device: temperature_sensor Super_Racer
2023-08-15 22:16:20,176 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-15 22:16:20,178 [main_menu.py:update_graph_visibility()] - Could not create graph tempstore: []
2023-08-15 22:16:20,179 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-15 22:16:20,201 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-15 22:16:28,657 [screen.py:_load_panel()] - Loading panel: print
2023-08-15 22:16:28,741 [print.py:__init__()] - 24h time is True
2023-08-15 22:16:28,797 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > print
2023-08-15 22:16:28,798 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2023-08-15 22:16:30,741 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-15 22:16:30,772 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-15 22:16:30,776 [main_menu.py:update_graph_visibility()] - Could not create graph tempstore: []
2023-08-15 22:16:30,777 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
2023-08-15 22:16:32,696 [screen.py:_go_to_submenu()] - #### Go to submenu config
2023-08-15 22:16:32,697 [screen.py:_go_to_submenu()] - #### Menu __main
2023-08-15 22:16:32,707 [screen.py:_load_panel()] - Loading panel: menu
2023-08-15 22:16:33,150 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > config
2023-08-15 22:16:33,208 [menu.py:arrangeMenuItems()] - X > config calibrate
2023-08-15 22:16:33,226 [menu.py:arrangeMenuItems()] - X > config camera
2023-08-15 22:16:33,246 [menu.py:arrangeMenuItems()] - X > config power
2023-08-15 22:16:34,515 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-15 22:16:34,536 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-15 22:16:34,540 [main_menu.py:update_graph_visibility()] - Could not create graph tempstore: []
2023-08-15 22:16:34,541 [base_panel.py:show_heaters()] - Couldn't create heaters box: 'heater_bed_box'
```

alfrix, 2023-08-15:
forgot to move a condition, so i pushed a fixup baabfc2e8077978e6a4f381050d71810eab6241c
i find very strange that moonraker takes so long to initialize the temp store

CIS94500, 2023-08-15:
I applied the baabfc2 patch and now the graph is displayed after a few seconds.

I have 2 instances of moonraker, it may be a lead for the abnormally long start..

I continue the tests.

THANKS :)

CIS94500, 2023-08-17:
I did some tests and the problem is solved, I let you close the ticket at your convenience.

commit 4800b564 (2023-08-18): refactor tempstore initialization

fixes #1074
closed 2023-08-18 by alfrix