KlipperScreen/KlipperScreen #1082: [FR] Layer Information during print

PatrickGlatz, 2023-08-16:
### Requested feature

During print, also show the current and the max. Layer, e.g.

Layer:  67/ 157

### Is your feature request related to a problem?

No, simply a small feature

### Describe alternatives you have considered

The height information is also helpfull, but it would be good to have the layer information (this helps me a lot more)

alfrix, 2023-08-16:
It's already possible:
https://klipperscreen.readthedocs.io/en/latest/Panels/Job_status/#position-info
https://klipperscreen.readthedocs.io/en/latest/Quicktips/#layer-progress

thinking in terms of layers it's flawed if you for example have a color change the layer number will change according to slicing layer height or if variable layer height is used, but the position in z will not change

closed 2023-08-16 by alfrix

PatrickGlatz, 2023-08-16:
Awesome, thanks, didn't now that this is already possible. Thanks for the hint.

The thing is, that i am matching curas layer preview with the current print layer. That is why i prefer layers instead of height. Thanks anyway