KlipperScreen/KlipperScreen #1086: [BUG] Relative moonraker webcam url doesn't work

sowa705, 2023-08-18:
### What happened?

When setting up a webcam in moonraker the documentation states that the url to the webcam stream can be relative: https://moonraker.readthedocs.io/en/latest/configuration/#webcam

KlipperScreen fails to open a relative webcam stream but works fine when the camera stream url is absolute

### What did you expect to happen instead?

Camera should work as relative urls work perfectly in mainsail

### How to reproduce this bug?

* Set a webcam stream url in moonraker.conf to `/webcam/?action=stream`
* Open camera on klipperscreen
* Error: file: Cannot open file '/webcam/?action=stream'

### Log output

```shell
sie 18 19:01:49 server KlipperScreen-start.sh[14926]: 2023-08-18 19:01:49,180 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > camera
sie 18 19:01:49 server KlipperScreen-start.sh[14926]: 2023-08-18 19:01:49,180 [camera.py:play()] - video filters: rotate:0.0
sie 18 19:01:49 server KlipperScreen-start.sh[14926]: 2023-08-18 19:01:49,211 [camera.py:play()] - Camera URL: /webcam/?action=stream
sie 18 19:01:49 server KlipperScreen-start.sh[14926]: 2023-08-18 19:01:49,222 [camera.py:log()] - [error] file: Cannot open file '/webcam/?action=stream': Nie ma takiego pliku ani katalogu
sie 18 19:01:49 server KlipperScreen-start.sh[14926]: 2023-08-18 19:01:49,224 [camera.py:log()] - [error] stream: Failed to open /webcam/?action=stream.
```

Guilouz, 2023-08-19:
Confirmed.

alfrix, 2023-08-19:
i don't have a camera so #1088 needs testing

sowa705, 2023-08-19:
I'm away from home until Monday, I can test when I get back.

Bazeone, 2023-08-19:
Confirmed.

o1hitman1o, 2023-08-20:
Setting stream_url and snapshot_url to http://(device ip)/webcam/?action=stream and snapshot_url: http://(device ip)/webcam/?action=snapshot gets the camera working in klipperscreen and working in Mainsail but there is a major delay. Setting stream_url and snapshot_url to /webcam/?action=stream and snapshot_url: /webcam/?action=snapshot fixes the camera in Mainsail but breaks it in klipperscreen.

PR #1088 (2023-08-21): camera: relative url support close #1086
also close #1092
commit c3ef1dd7 (2023-08-21): camera: relative url support close #1086 (#1088)

close #1092
closed 2023-08-21 by alfrix