KlipperScreen/KlipperScreen #1087: [BUG] Icons stay busy on Custom Panels

Guilouz, 2023-08-19:
### What happened?

When creating customs panels in Klipperscreen.conf file or default.conf, all icons stay busy after they are pressed.

Exit from current panel and re-enter in custom panel, icons are go back.

[image]

### What did you expect to happen instead?

Icons not stay busy on custom panels

### How to reproduce this bug?

Just create a custom panel and try

### Log output

```shell
2023-08-19 02:24:54,873 [screen.py:update_size()] - Size changed: 1024x600
2023-08-19 02:24:54,963 [screen.py:_load_panel()] - Loading panel: main_menu
2023-08-19 02:24:55,022 [main_menu.py:__init__()] - ### Making MainMenu
2023-08-19 02:24:55,028 [main_menu.py:add_device()] - Adding device: extruder
2023-08-19 02:24:55,034 [main_menu.py:add_device()] - Adding device: heater_bed
2023-08-19 02:24:55,040 [main_menu.py:add_device()] - Adding device: temperature_sensor BigTreeTech_Pad7
2023-08-19 02:24:55,047 [main_menu.py:add_device()] - Adding device: temperature_sensor Motherboard
2023-08-19 02:24:55,079 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-08-19 02:25:04,051 [screen.py:_go_to_submenu()] - #### Go to submenu actions
2023-08-19 02:25:04,051 [screen.py:_go_to_submenu()] - #### Menu __main
2023-08-19 02:25:04,056 [screen.py:_load_panel()] - Loading panel: menu
2023-08-19 02:25:04,271 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > actions
2023-08-19 02:25:04,290 [menu.py:arrangeMenuItems()] - X > actions power
2023-08-19 02:25:06,123 [screen.py:_go_to_submenu()] - #### Go to submenu actions neopixels
2023-08-19 02:25:06,123 [screen.py:_go_to_submenu()] - #### Menu __main
2023-08-19 02:25:06,127 [screen.py:_load_panel()] - Loading panel: menu
2023-08-19 02:25:06,300 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > actions > actions neopixels
2023-08-19 02:25:07,481 [screen.py:_send_action()] - printer.gcode.script: {'script': 'NEOPIXEL_ON'}
2023-08-19 02:25:10,111 [screen.py:_send_action()] - printer.gcode.script: {'script': 'NEOPIXEL_BLUE'}
2023-08-19 02:25:12,386 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-19 02:25:12,411 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > actions
2023-08-19 02:25:12,432 [menu.py:arrangeMenuItems()] - X > actions power
2023-08-19 02:25:14,272 [screen.py:_go_to_submenu()] - #### Go to submenu actions neopixels
2023-08-19 02:25:14,272 [screen.py:_go_to_submenu()] - #### Menu __main
2023-08-19 02:25:14,299 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > actions > actions neopixels
2023-08-19 02:25:15,093 [screen.py:_send_action()] - printer.gcode.script: {'script': 'NEOPIXEL_ON'}
2023-08-19 02:25:15,838 [screen.py:_menu_go_back()] - #### Menu go back
2023-08-19 02:25:15,863 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > actions
2023-08-19 02:25:15,883 [menu.py:arrangeMenuItems()] - X > actions power
```

Guilouz, 2023-08-21:
I just saw that it also appears on other panels like move (homig stay busy for exempale) and extrude panels.

Guilouz, 2023-08-21:
@alfrix This is this commit that cause issue: https://github.com/KlipperScreen/KlipperScreen/commit/90fc3288d77c52b1b5b42ce6978451742d04eea7#diff-4dc6710f3893ed193fc05caa325c9fe5b5d9e32e3bf2c73ae725a7dfbc1c9cd1

This definitions:

```
def _send_action(self, widget, method, params):
        logging.info(f"{method}: {params}")
        self._ws.send_method(method, params)
        if type(widget) == Gtk.Button:
            self.gtk.Button_busy(widget, True)
            self._ws.send_method(method, params, self.enable_widget, widget)
        else:
            self._ws.send_method(method, params)

    def enable_widget(self, *args):
        for x in args:
            if type(x) == Gtk.Button:
                GLib.timeout_add(150, self.gtk.Button_busy, x, False)
```

This is also what causes the problem of the error popup which is displayed for half a second or which is displayed only when leaving the current panel.

alfrix, 2023-08-21:
the button is supposed to stay busy until the callback is returned, meaning if you press 'home all' or whatever that take a long time and then anything else, it will remain busy until the action is performed

not having this issue, please always test with this repo and not your fork, since you may have differences in your fork

Guilouz, 2023-08-21:
@alfrix My fork is the same, if I revert back this commit, it's working as expected.

Guilouz, 2023-08-21:
To be sure is not my fork, I just added and the issue is the same on Custom Panels, there is really something wrong with that:

In screen.py:

```
    def _send_action(self, widget, method, params):
        logging.info(f"{method}: {params}")

        if type(widget) == Gtk.Button:
            self.gtk.Button_busy(widget, True)
            self._ws.send_method(method, params, self.enable_widget, widget)
        else:
            self._ws.send_method(method, params)

    def enable_widget(self, *args):
        for x in args:
            if type(x) == Gtk.Button:
                GLib.timeout_add(150, self.gtk.Button_busy, x, False)
```

and in KlippyGtk.py:
```
    def Button(self, image_name=None, label=None, style=None, scale=None, position=Gtk.PositionType.TOP, lines=2):
        if self.font_size_type == "max" and label is not None and scale is None:
            image_name = None
        b = Gtk.Button()
        if label is not None:
            b.set_label(label.replace("\n", " "))
        b.set_hexpand(True)
        b.set_vexpand(True)
        b.set_can_focus(False)
        if image_name is not None:
            if scale is None:
                scale = self.button_image_scale
            if label is None:
                scale = scale * 1.4
            width = height = self.img_scale * scale
            b.set_image(self.Image(image_name, width, height))
            b.set_image_position(position)
            b.set_always_show_image(True)
            spinner = Gtk.Spinner.new()
            spinner.set_no_show_all(True)
            spinner.set_size_request(width, height)
            spinner.hide()
            b.get_child().get_child().add(spinner)

    @staticmethod
    def Button_busy(widget, busy):
        box = widget.get_child().get_child()
        if busy:
            widget.set_sensitive(False)
            widget.set_always_show_image(False)
            box.get_children()[0].hide()
            if type(box.get_children()[1]) == Gtk.Spinner:
                box.get_children()[1].start()
                box.get_children()[1].show()
        else:
            box.get_children()[0].show()
            if type(box.get_children()[1]) == Gtk.Spinner:
                box.get_children()[1].stop()
                box.get_children()[1].hide()
            widget.set_always_show_image(True)
            widget.set_sensitive(True)
```

alfrix, 2023-08-21:
excellent because i couldn't reproduce the issue at all

Guilouz, 2023-08-21:
@alfrix I have find the issue it's happen when a custom background is set for theme.

Like this:

list row,
treeview.view,
window {
    background-color: #13181C;
    background-image: url("./styles/z-bolt/background.svg");
    background-size: cover;
}

If I remove background it's working.

alfrix, 2023-08-22:
no it doesn't make it happen, and you must separate the window into it's own block...
perhaps you should chase that error popup, it may shine some clues

Guilouz, 2023-08-22:
What do you mean by separate window into it’s own block?

alfrix, 2023-08-26:
i did some refactoring to make the function safer in f77e382 it should fix this

Guilouz, 2023-08-26:
> i did some refactoring to make the function safer in [f77e382](https://github.com/KlipperScreen/KlipperScreen/commit/f77e38225208298654d259d2d7d0924c415e7ba0) it should fix this

Just try, that does not fix issue when background is enabled.

alfrix, 2023-08-26:
[Screencast_20230826_165938.webm](https://github.com/KlipperScreen/KlipperScreen/assets/1247237/31d30ebf-9e5b-44bd-a446-6beadb335722)

Guilouz, 2023-08-26:
I don't understand where this can come from, if I removed this commit all is working as expected : https://github.com/KlipperScreen/KlipperScreen/commit/90fc3288d77c52b1b5b42ce6978451742d04eea7#diff-4dc6710f3893ed193fc05caa325c9fe5b5d9e32e3bf2c73ae725a7dfbc1c9cd1

alfrix, 2023-08-28:
> I don't understand where this can come from, if I removed this commit all is working as expected : [90fc328#diff-4dc6710f3893ed193fc05caa325c9fe5b5d9e32e3bf2c73ae725a7dfbc1c9cd1](https://github.com/KlipperScreen/KlipperScreen/commit/90fc3288d77c52b1b5b42ce6978451742d04eea7#diff-4dc6710f3893ed193fc05caa325c9fe5b5d9e32e3bf2c73ae725a7dfbc1c9cd1)
> 
> Can you try my fork available here (it's a fork for flsun speeder pad with some changes for delta printers): https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad.git
> 
> (Needed to re-enable background of themes to see issue)

your fork seems to work correctly with and without the background, i've not encountered this issue, but you have some warnings:
```
Gtk-WARNING **: 00:45:43.165: Theme parsing error: <data>:707:19: Not using units is deprecated. Assuming 'px'.
Gtk-WARNING **: 00:45:43.165: Theme parsing error: <data>:808:19: Not using units is deprecated. Assuming 'px'.
Gtk-CRITICAL **: 00:45:44.240: gtk_widget_set_size_request: assertion 'height >= -1' failed
```

alfrix, 2023-08-28:
[Screencast_20230828_005210-1.webm](https://github.com/KlipperScreen/KlipperScreen/assets/1247237/08dcb926-9965-485e-b021-89182c3569ec)

Guilouz, 2023-08-28:
So what can cause this ? I have tested on Flsun Speeder Pad (1024x600) that is on Ubuntu and a BTT Pad (1024x600) that is on MainsailOS.

alfrix, 2023-08-28:
> So what can cause this ? I have tested on Flsun Speeder Pad (1024x600) that is on Ubuntu and a BTT Pad (1024x600) that is on MainsailOS.

the gtk version perhaps?, you will have to place logging lines in the different functions to see where it stops executing, i'm guessing that enable_widget is not being called or there is an issue with the timeout function

Guilouz, 2023-08-28:
> the gtk version perhaps?, you will have to place logging lines in the different functions to see where it stops executing, i'm guessing that enable_widget is not being called or there is an issue [...]

I have same issue if I use your repo with a theme with a background.

alfrix, 2023-09-23:
closing since this does not happen on the mainline repo, i haven't been able to reproduce the issue at all, it has not been reported, and my directions to use log files to try to trackdown the issue were ignored

cdsmith69, 2023-11-16:
I am experiencing this issue as well.  I am on the mainline repo, but I have my own theme with a background image.

The problem appears when I have a background image and have the style.css file written to center and "cover" the background image.

My company is developing a couple Klipper based printers.  One has a 5 inch 800x480 HDMI touchscreen.  The other has a 7 inch 1024x600 HDMI touchscreen.  I made a custom theme where I changed a bunch of colors, added a background image, and swapped a few icon files.

I made the background image 800x480 and called it out in the style.css file with 
window {background-image: url("/home/pi/KlipperScreen/styles/LulzBot/Lulzbot_Logo_Very_Dark.jpg");}

This works fine on the printer with the 5 inch screen.  But when I put the theme on the printer with the 7 inch screen I decided to get clever and instead of using an image editor to resize it to 1024x600, I added a couple lines to the style.css to center and resize the image like this:

window {background-image: url("/home/pi/KlipperScreen/styles/LulzBot/LulzBot_Logo_Very_Dark.jpg");
                          background-position: center;
                          background-size: cover;}

This is what "breaks" the busy icon.  With "background-size: cover;" in there, the busy icon does not go away, I have to back out of the menu and go back in.  Or on the extrude screen, the icons will come back if I touch a different distance or speed number.  

I removed the "background-size: cover;" on the printer with the 7 inch screen, and the issue went away.  Having "background-position: center;" in appears to be OK.  I don't see the issue with center in but not cover.

I will fix my printers by resizing the background image to 1024x600 for the printer with 7 inch screen and remove the center and cover.

Guilouz, 2023-11-16:
So there is definitely a problem, I wasn't crazy 😆

alfrix, 2023-12-12:
i will reopen, but still unable to reproduce the issue even with the css tweaks

Guilouz, 2023-12-12:
@alfrix I confirm it’s background-size: cover the issue with large screen (7 inch screens) when removed et background set in 1024x600 it’s working.

alfrix, 2023-12-30:
no matter what i do, i still can't reproduce this at all, i suspect that it has to do with the hw used

can anyone with the issue try this:
https://github.com/alfrix/KlipperScreen/commit/7c61e3aeb37595c70f8b8bf0472374c84f1bbff1

and post the log result?

cdsmith69, 2024-01-02:
The hardware I am using is:

OS: Raspbian GNU/Linux 10 (buster) armv7l
Host: Raspberry Pi 3 Model B Plus Rev 1.3
Kernel: 5.10.103-v7+
Connected to a Big Tree Tech Octopus board and Big Tree Tech HDMI7 screen.

I edited my style.css file back to using the 800x480 image and the center and cover settings, as in my Nov 16 comment, and verified that the issue came back.

I then hand edited the changes to those lines of code in the above commit, ran a couple home commands, and saved the log file.
I didn't think to roll over the log first, and it was huge, so I hand trimmed off and saved the end from where I found the last "KlipperScreen Log Start."
[KlipperScreen.trimmed.log](https://github.com/KlipperScreen/KlipperScreen/files/13811901/KlipperScreen.trimmed.log)

Hopefully this is helpful.

alfrix, 2024-01-03:
> The hardware I am using is:
>
> OS: Raspbian GNU/Linux 10 (buster) armv7l Host: Raspberry Pi 3 Model B Plus Rev 1.3 Kernel: 5.10.103-v7+ Connected to a Big Tree Tech Octopus board and Big Tree Tech [...]

```
2024-01-02 08:54:19,918 [screen.py:enable_widget()] - {'jsonrpc': '2.0', 'result': 'ok', 'id': 104}
2024-01-02 08:54:19,918 [screen.py:enable_widget()] - printer.gcode.script
2024-01-02 08:54:19,919 [screen.py:enable_widget()] - {'script': 'G91\nG0 Z+25 F600'}
2024-01-02 08:54:19,919 [screen.py:enable_widget()] - <Gtk.Button object at 0x63e96648 (GtkButton at 0xaa3120)>
```
the callback is working, the enable function is being called, the button is there...
maybe the issue is in the button_busy function, or it's on the old versions of everything used by the outdated OS
have you tested debian 12 or 11 ?

and python 3.7 is eol  and most packages are dropping support, i will start giving users warnings in the interface shortly

cdsmith69, 2024-02-01:
Finally had some time, so I built a whole new card for my printer with the latest Raspberry Pi OS, then used KIAUH to reinstall everything, then put back all my same config files.  So now I'm running:
OS: Raspbian GNU/Linux 12 (bookworm) armv7l
Host: Raspberry Pi 3 Model B Plus Rev 1.3
Kernel: 6.1.0-rpi8-rpi-v7
Python 3.11.2

I still see the issue if I resize a background image with background-position: center; and background-size: cover;

I don't know if there is something in my configuration that is causing this, maybe the changes I made to the CSS files for my theme, but I am not very worried about this issue anymore.  We have a workaround, just resize the background image to fit the screen and don't use center and cover.

l0ud, 2024-02-08:
I have the same exact issue on Raspberry Pi 3B+ using this theme bundle: https://github.com/bumbeng/Klipperscreen_blurry_theme which happens to use background image and transparency.

Issue happens every time on blurry, blurry2 themes and sporadically on "simple" theme.
"Print" page is also affected with infinite "Loading..." (after going back and trying again it loads correctly) and right top icon spinning indefinitely there.

I think this is a timing issue, when GUI don't update or redraw fast enough to be ready to accept callback - just guessing. It is pronounced on low-end hardware - I use Raspberry Pi 3B+ and Chinese crappy Klipper pads are probably not better hardware-wise. It's probably much harder (impossible?) to reproduce on a stronger Raspberry Pi 4 or PC. Setting background image and/or transparency on objects makes whole thing render much slower. Probably slow enough to trigger that issue.

l0ud, 2024-02-09:
Alright, so after delving deep into this issue, I've found the root cause.

Essentially, when Gtk.Spinner is animating, the CPU gets so busy that stuff scheduled with GLib.idle_add is never getting executed.
Hiding a spinner and making button active is usually done by callback provided to KlippyWebsocket, which is executed by GLib.idle_add. As spinner animation is playing, it's preventing execution of callback intended to stop it, leading to a button deadlock.

The print menu is similarly affected as it's using GLib.idle_add to queue up the display of list while showing a spinner on top-right corner.

A quick, albeit not always reliable, workaround involves limiting spinner's fps in css:
`spinner { animation-timing-function: steps(12); }`

Even with this adjustment, on my Raspberry Pi, I can still manage to "lock up" buttons by triggering multiple spinners simultaneously—simply rapidly pressing X-, Y-, X+ buttons in the Move menu is sufficient to max out the CPU.

For a more permanent solution, I've replaced all occurrences of:
`GLib.idle_add(`
to
`GLib.timeout_add(0, `

The only noticeable side effect I've observed is that the "Loading..." message in the "Print" screen no longer appears.

alfrix, 2024-02-09:
> I have the same exact issue on Raspberry Pi 3B+ using this theme bundle: https://github.com/bumbeng/Klipperscreen_blurry_theme which happens to use background image and transparency.
>
> Issue happens [...]

thank you, finally that blurry2 theme triggers the issue for me

> For a more permanent solution, I've replaced all occurrences of:
GLib.idle_add(
to
GLib.timeout_add(0, 

timeouts are very quirky, not recommended, i just tested and the most consistent fix is elevating the priority of the callback

PR #1266 (2024-02-18): ws: fix issue that made buttons busy forever fixes #1087
fixes #1087 
i want to do more testing before merging
if someone tests this, please report
commit d9a361ca (2024-02-18): ws: fix issue that made buttons busy forever fixes #1087 (#1266)

* ws: fix issue that made buttons busy forever fixes #1087

* ws: reduce to high_idle due to touch not as responsive

* ws: make string translatable

* less animation steps for the spinner

the slower spinning gear icon is actually more pleasant to look
closed 2024-02-18 by alfrix