KlipperScreen/KlipperScreen #1093: Main Printer

mpx1986, 2023-08-21:
### What happened?

I don´t need this option. Many also need this or would deactivate this.

### What did you expect to happen instead?

I don´t need this option. Many also need this or would deactivate this.

### How to reproduce this bug?

I don´t need this option. Many also need this or would deactivate this.

### Log output

```shell
I don´t need this option. Many also need this or would deactivate this.
```

mpx1986, 2023-08-21:
Please make a option to deactivate the default printer

alfrix, 2023-08-21:
the bug report form exists for a reason never copy paste everything like this, explain yourself.
the form guides you, the only reason why i don't close this, it's because someone else have described the issue in discord

alfrix, 2023-08-21:
fixed the bug report, try to follow the form next time

commit 353bdc73 (2023-08-21): base_panel: fix printer select fixes #1093
closed 2023-08-21 by alfrix