KlipperScreen/KlipperScreen #1097: [FR] Scan QR Code Button

NonaSuomy, 2023-08-22:
### Requested feature

Hi how would you add an overlayed button to the camera.py linked to my macro that scans a QR code on the camera to tell spoolman which spool to use?

Macro is setup like this:

Installs ZBar
```
mkdir code
cd code
sudo apt install autopoint build-essential libv4l-dev libtool
git clone https://github.com/mchehab/zbar.git
cd zbar
autoreconf -vfi
./configure
sudo make install
```

Install qrencode
`sudo apt install qrencode`

Generate a QRCode
```
qrencode -s 6 -l H -o "qrcodespool1.png" "web+spoolman:s-1"
```

On your 2D printer, print this on a sticker for your spool.

CLI
```
cd /home/pi/klipper/klippy/extras/
wget https://raw.githubusercontent.com/th33xitus/kiauh/master/resources/gcode_shell_command.py
```

Change `<PrinterIP>` to the IP of your 3D printer below.

mainsail.cfg
```
[gcode_shell_command qrcode_qrscanner]
command: sh /home/pi/klipper_config/qrcodespoolman.sh
verbose: True
timeout: 2.
[gcode_macro QRCode_Scan]
gcode:
    RUN_SHELL_COMMAND CMD=qrcode_qrscanner
```
/home/pi/klipper_config/qrcodespoolman.sh
```bash
#!/bin/sh
# NonaSuomy PoC scan QRCodes via 3D printer camera
wget http://<PrinterIP>/webcam/?action=snapshot -O /home/pi/printer_data/gcodes/qrcode.jpg
SPOOLID=$(zbarimg -S*.enable /home/pi/printer_data/gcodes/qrcode.png | sed 's/[^0-9]*\([0-9]\+\).*/\1/')
echo $SPOOLID
curl -X POST -H "Content-Type: application/json" -d "{\"spool_id\": \"$SPOOLID\"}" http://<PrinterIP>:7125/server/spoolman/spool_id
```

Put your spool qrcode upto the chamber camera and hit "QRCode Scan" on the dashboard of MainsailOS.

[image]

http://<PrinterIP:7125/server/spoolman/spool_id
```
{"result": {"spool_id": 1}}
```

For now I’ll be using the run shell cmd just to get this PoC working but would you happen to also know of a way to accomplish this without using run shell cmd? Apparently mainsail frowns upon its usage.

Thank you

### Is your feature request related to a problem?

No

NonaSuomy, 2023-12-13:
Any idea how to accomplish putting the macro button over the video feed?

cdedwards, 2025-02-28:
on debian, you need to install zbar-tools in order for this to work.

NonaSuomy, 2025-08-15:
Yay! Scan with chamber cam...
[image]

Selects correct roll...
[image]