KlipperScreen/KlipperScreen #1104: [BUG] MACROS are no longer being displayed after updates

R4toLoko, 2023-08-31:
### What happened?

Unable to `UNLOAD_FILAMENT` and `LOAD_FILAMENT` during printing after a printer pause, or invoke any macro for the matter.
The macros screen was completely empty ( It is expected to don't display all the other macros ) and this happened after the latest update which I checked now and there was a new release available and that didn't fix the problem it created.

`Version = v0.3.5-16`

I was collecting the log to send here and this is so wrong, I never set those macros to `false`

[image]

I even tried to set them to true to force them being displayed but nope:

[image]

The workaround I found was to create a

`X_LOAD_FILAMENT` and `X_UNLOAD_FILAMENT` to call `LOAD_FILAMENT` and `UNLOAD_FILAMENT` respectively and that works but this is wrong.

[image]

During the print last night, I had to grab my phone to call those macros via Mainsail > Console which is beyond madness.

### What did you expect to happen instead?

Be able to hit `UNLOAD_FILAMENT` and `LOAD_FILAMENT` during printing after a printer pause like I have been doing for ages.
I mean, being able to use these 2 macros and only them are having this problem.

### How to reproduce this bug?

1. Update everything
2. Slice a print with pause at layer X
3. Try to access the macros menu to UNLOAD and LOAD the filament.

### Log output

```shell
=======================
2023-08-30 12:19:20,755 [config.py:__init__()] - Configured printers: [
  {
    "Trident": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-08-30 12:19:20,773 [config.py:create_translations()] - Selected lang: None OS lang: en_AU
2023-08-30 12:19:20,773 [config.py:install_language()] - Using system lang
2023-08-30 12:19:20,774 [config.py:install_language()] - Using lang en
2023-08-30 12:19:20,778 [screen.py:__init__()] - Screen resolution: 1024x600
2023-08-30 12:19:20,778 [KlippyGtk.py:__init__()] - Font size: 20.7 (medium)
2023-08-30 12:19:20,986 [screen.py:set_screenblanking_timeout()] - Screen blanking: off
2023-08-30 12:19:20,989 [screen.py:initial_connection()] - Default printer: Trident
2023-08-30 12:19:20,989 [screen.py:connect_printer()] - Connecting to printer: Trident
2023-08-30 12:19:20,989 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-08-30 12:19:21,000 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-08-30 12:19:21,002 [KlippyWebsocket.py:connect()] - Attempting to connect
2023-08-30 12:19:21,005 [KlippyRest.py:_do_request()] - host=127.0.0.1, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2023-08-30 12:19:21,010 [screen.py:update_size()] - Size changed: 1024x600
2023-08-30 12:19:31,510 [KlippyWebsocket.py:connect()] - Attempting to connect
2023-08-30 12:19:31,520 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-30 12:19:31,528 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2023-08-30 12:19:31,529 [KlippyWebsocket.py:connect()] - Starting websocket thread
2023-08-30 12:19:31,535 [_logging.py:info()] - Websocket connected
2023-08-30 12:19:31,535 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2023-08-30 12:19:31,543 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-30 12:19:31,544 [base_panel.py:set_ks_printer_cfg()] - Titlebar name type: None items: ['chamber', 'MCU', 'beacon']
2023-08-30 12:19:31,551 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-08-30 12:19:31,552 [screen.py:init_printer()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'timelapse', 'update_manager', 'octoprint_compat', 'authorization', 'history'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'timelapse', 'timelapse_frames', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-138-gfe12095', 'missing_klippy_requirements': [], 'api_version': [1, 3, 0], 'api_version_string': '1.3.0'}
[... 401 lines cut ...]
2023-08-30 19:31:54,443 [screen.py:_load_panel()] - Loading panel: menu
2023-08-30 19:31:54,519 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more
2023-08-30 19:31:54,520 [menu.py:arrangeMenuItems()] - X > more bedlevel
2023-08-30 19:31:54,522 [menu.py:arrangeMenuItems()] - X > more retraction
2023-08-30 19:31:54,525 [menu.py:arrangeMenuItems()] - X > more power
2023-08-30 19:31:54,529 [menu.py:arrangeMenuItems()] - X > more spoolman
2023-08-30 19:31:55,751 [screen.py:_load_panel()] - Loading panel: system
2023-08-30 19:31:55,775 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/machine/update/status
2023-08-30 19:31:55,777 [system.py:_already_updated()] - beacon v1.0.0-6
2023-08-30 19:31:55,780 [system.py:_needs_update()] - klipper v0.11.0-257 -> v0.11.0-271
2023-08-30 19:31:55,781 [system.py:_already_updated()] - mainsail v2.7.1
2023-08-30 19:31:55,785 [system.py:_already_updated()] - moonraker v0.8.0-138
2023-08-30 19:31:55,786 [system.py:_needs_update()] - system  -> 2
2023-08-30 19:31:55,787 [system.py:_needs_update()] - timelapse v0.0.1-141 -> v0.0.1-142
2023-08-30 19:31:55,796 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > system
2023-08-30 19:31:55,800 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/machine/update/status?refresh=false
2023-08-30 19:31:55,800 [system.py:_already_updated()] - beacon v1.0.0-6
2023-08-30 19:31:55,801 [system.py:_needs_update()] - klipper v0.11.0-257 -> v0.11.0-271
2023-08-30 19:31:55,802 [system.py:_already_updated()] - mainsail v2.7.1
2023-08-30 19:31:55,802 [system.py:_already_updated()] - moonraker v0.8.0-138
2023-08-30 19:31:55,802 [system.py:_needs_update()] - system  -> 2
2023-08-30 19:31:55,803 [system.py:_needs_update()] - timelapse v0.0.1-141 -> v0.0.1-142
2023-08-30 19:31:56,844 [KlippyGtk.py:Dialog()] - Showing dialog <Gtk.Dialog object at 0x7f5ab466b540 (GtkDialog at 0x1f2d940)>
2023-08-30 19:31:58,093 [KlippyGtk.py:remove_dialog()] - Removing Dialog
```

R4toLoko, 2023-08-31:
Something is very wrong for some time now, even with all on this in place, the display keeps going blank which is driving me nuts

```
#~# [main]
#~# theme = material-darker
#~# screen_blanking = off
#~# use_dpms = False
#~# autoclose_popups = False
#~# show_heater_power = True
#~# print_sort_dir = date_desc
```

This is not a Debian problem, everything was just fine and out of the blue anything energy saving I disable does nothing.

alfrix, 2023-08-31:
The removal of load and unload macros from the macro panel was done on  purpose, the idea is force the use of the extrude panel in which the buttons will disable if the printer is printing which is safer.
the part of the macro panel being blank is something to consider

fix: if no macro is valid to show, then do not show the shortcut
i am also considering the suggestion someone made about auto-opening the extrude panel on pause, which is probably what you want here

R4toLoko, 2023-08-31:
> The removal of load and unload macros from the macro panel was done on purpose, the idea is force the use of the extrude panel in which the buttons will disable if the printer is printing which is [...]

I see, it does make sense in theory but not in practice with 2 major problems.

**Problem 1:**
To load the filament I was using the extrude panel which is painful slow, it would take easily over a minute, easily.
With `LOAD_FILAMENT` and `UNLOAD_FILAMENT` takes what ~10-15s in total.

**Problem 2:**
The unload does exactly that, unload only.
There is no quick purge, so many times everything was stuck not purging at all with the filament stuck like if I was doing cold pull.
I had to manually hit load which again, is slow, and then hit unload which is also slow leaving a huge string all the way to the extruder gear.
A solution replacing another, must make that process simpler and better, not worse.

> the idea is force the use of the extrude panel in which the buttons will disable if the printer is printing which is safer.

I get the point but nobody will be executing such macros when the printer is running, I mean, there is absolutely zero reasons to do that.

> the part of the macro panel being blank is something to consider

It was done by me and expected, I disabled all the other macros coz they were driving me nuts, LOAD and UNLOAD is all I care. This is expected.

--- 

> fix: if no macro is valid to show, then do not show the shortcut
> i am also considering the suggestion someone made about auto-opening the extrude panel on pause, which is probably what you want here

This sounds beautiful but only if we can speed up the load and unload, over a minute to get this simple operation done is madness haha
Perhaps it will allow us to increase the speed and distance and save it so the next time I hit load/unload, I don't need to tweak those values every-time.

These are my LOAD/UNLOAD at the moment:

```
[gcode_macro LOAD_FILAMENT]
gcode:
    SAVE_GCODE_STATE NAME=load_state
    G91
    G92 E0
    G1 E101 F300 # fast-load
    RESTORE_GCODE_STATE NAME=load_state
```

unload must do that purge first which it doesn't, not when I tested it.
If the fix will have this purge/unload logic and allow us to set the speed and distance, boy, solid stuff right there.

```
[gcode_macro UNLOAD_FILAMENT]
gcode:
    SAVE_GCODE_STATE NAME=unload_state
    G91
    G92 E0
    G1 E25 F1000 # purge
    G1 E-101 F1500 # fast-unload
    RESTORE_GCODE_STATE NAME=unload_state
```
As for now, I will keep using the `X_LOAD_FILAMENT` calling  `LOAD_FILAMENT` for example

Thank you so much for the reply.

alfrix, 2023-09-01:
> Problem 1:
To load the filament I was using the extrude panel which is painful slow, it would take easily over a minute, easily.
With LOAD_FILAMENT and UNLOAD_FILAMENT takes what ~10-15s in total.

the extrude panel has both load and unload buttons and they run LOAD_FILAMENT and UNLOAD_FILAMENT so your point doesn't make any sense, it should be the same

> Problem 2:
The unload does exactly that, unload only.
There is no quick purge...

Again the unload button in the extrude panel runs the macro so just modify the macro to your liking 

Reference
[image]

here are the example macros:
https://klipperscreen.readthedocs.io/en/latest/macros/#load_filament-unload_filament

R4toLoko, 2023-09-01:
> the extrude panel has both load and unload buttons and they run LOAD_FILAMENT and UNLOAD_FILAMENT so your point doesn't make any sense, it should be the same
>
>
> Again the unload button in the extrude [...]

That is weird, what you are saying is totally different from the experience I had, that is why I never used it again like ever.
Got it, so that LOAD/UNLOAD calls the macros I have instead of having their own logic, beautiful.

Will give that a go then.

Thank you

closed 2023-09-01 by R4toLoko

NiiXXiiN, 2023-09-17:
I have the exact same issue with @R4toLoko . 
I believe this is a step back, why should we tinker with extra macros if we have already set up load and unload routines globally for the printer?
I disagree with the decision and for my usecase is quite the hassle. 
These are my macros which don't work through klipperscreen.

```
[gcode_macro UNLOAD_FILAMENT]
gcode:
	SAVE_GCODE_STATE NAME=unload_state
	G91
	{% if params.TEMP is defined or printer.extruder.can_extrude|lower == 'false' %}
		M117 Heating...
		# Heat up hotend to provided temp or 220 as default as that should work OK with most filaments.
		M104 S{params.TEMP|default(220, true)}
		TEMPERATURE_WAIT SENSOR=extruder MINIMUM={params.TEMP|default(220, true)}
	{% endif %}
	{% set unload_speed = printer["gcode_macro RatOS"].filament_unload_speed|float * 60 %}
	{% set unload_length = printer["gcode_macro RatOS"].filament_unload_length|float %}
	M117 Unloading filament...
	# Extrude a bit
	G0 E10 F300
	# Extract filament to cold end area 
	G0 E-5 F3600
	# Wait for three seconds
	G4 P3000
	# Push back the filament to smash any stringing 
	G0 E5 F3600
	# Extract back fast in to the cold zone 
	G0 E-15 F3600
	# Continue extraction slowly, allow the filament time to cool solid before it reaches the gears       
	G0 E-{unload_length} F{unload_speed}
	M117 Filament unloaded!
	RESPOND MSG="Filament unloaded! Please inspect the tip of the filament before reloading."
	RESTORE_GCODE_STATE NAME=unload_state
```

```
[gcode_macro LOAD_FILAMENT]
gcode:
	SAVE_GCODE_STATE NAME=load_state
	G91
	# Heat up hotend to provided temp or 220 as default as that should work OK with most filaments.
	{% if params.TEMP is defined or printer.extruder.can_extrude|lower == 'false' %}
		M117 Heating...
		M104 S{params.TEMP|default(220, true)}
		TEMPERATURE_WAIT SENSOR=extruder MINIMUM={params.TEMP|default(220, true)}
	{% endif %}
	{% set load_speed = printer["gcode_macro RatOS"].filament_load_speed|float * 60 %}
	{% set load_length = printer["gcode_macro RatOS"].filament_load_length|float %}
	M117 Loading filament...
	# Load the filament into the hotend area.
	G0 E{load_length} F{load_speed}
	# Wait a secod
	G4 P1000
	# Purge
	G0 E40 F100
	# Wait for purge to complete
	M400
	M117 Filament loaded!
	RESPOND MSG="Filament loaded!"
	RESTORE_GCODE_STATE NAME=load_state
```

R4toLoko, 2023-09-17:
@NiiXXiiN ,

We need to start using like @alfrix mentioned.
The biggest mistake imo was removing them knowing that many people use that way without providing any solution like bringing the nozzle screen up, that was a big fail.

I haven't printed much with pause but when I did, instead of going to the macros menu, you go to the nozzle menu and the load and unload should invoke `LOAD_FILAMENT` and `UNLOAD_FILAMENT` in a more prettier way.

It has been working fine but like I said, before removing a feature, the replacement must be introduced FIRST, not before.

Give it a go and you should have no problems.