KlipperScreen/KlipperScreen #1113: [FR] All pauses should not open the extrude menu

rednaxal, 2023-09-04:
### Requested feature

Since this commit [https://github.com/KlipperScreen/KlipperScreen/commit/a6866e8a0fe643af0f1cc92ce179f0940a7e5fb0](https://github.com/KlipperScreen/KlipperScreen/commit/a6866e8a0fe643af0f1cc92ce179f0940a7e5fb0) all pauses started to open the extrude menu. Which makes no sense for many people. Because not all pauses do need the manual filament swap. Many pauses are being made for putting nuts, magnets, etc. While for filament swaps many people do rely on slicers' swaps and don't need to manually load/unload filaments through menu. 
For example I do all of my filament swaps through PrusaSlicer's built-in functionality. And upon filament swap all I need to do on the screen is to hit the "Resume" button. 

### Describe alternatives you have considered

If that commit saves someone's life then please add a setting for that behavior so that it doesn't makes things more complicated for others who don't need to load/unload filament upon each pause.

alfrix, 2023-09-04:
> ... Which makes no sense for many people ... many people do rely on slicers' swaps and don't need to manually load/unload ...

'many people' i wonder if you have statistics, because i'm sure i don't.

> all I need to  do on the screen is to hit the "Resume" button

if it's that good then place a `RESUME` command in the end of the load procedure

closed 2023-09-05 by alfrix

rednaxal, 2023-09-05:
> 'many people' i wonder if you have statistics, because i'm sure i don't.
Statistics of what? And why do we need a statistics? I know a LOT of people including myself who are using the pause function to put nuts and magnets in their prints.

> if it's that good then place a `RESUME` command in the end of the load procedure
My unload/load procedure is processed fully by PrusaSlicer with all of its ramming, wipe tower, etc. All I have to do during a pause for a filament swap is to remove the old filament end and put the new filament end to the extruder gears and only then hit the resume button manually. I.e. the resume can not be automatic. Neither the "Extrude" menu makes sense here. And I'm sure I'm not the only one who is using these features of PrusaSlicer.

alfrix, 2023-09-05:
>  I know a LOT of people including myself

if there are no statistics or telemetry you are making assumptions about quantities, you don't know if you and the people you know are a representative sample of the users, that is the point

>  wipe tower

nobody that i know uses this. withot a fully automated swap procedure (ERCF or others).

if you end up being there to press buttons or do stuff then a wipe tower doesn't make sense to me, i would load the filament manually to avoid the waste of material,  good loading procedure doesn't take a lot of time

it's sad so many people are against telemetry, it does give a lot of insight of what the majority of people wants, instead of a few loud voices

rednaxal, 2023-09-05:
> if there are no statistics or telemetry you are making assumptions about quantities, you don't know if you and the people you know are a representative sample of the users, that is the point

Well you wrote in your commit message "_most (maybe all pauses) are done for filament swaps_" - which seems like you really do have some statistics. While I'm not making any assumptions and strong statements as you can actually see. I'm just stating that MANY people are using pauses for other than filament change purposes. Whether that is most of the people or not - I have no idea. 

> nobody that i know uses this. withot a fully automated swap procedure (ERCF or others).
>
> if you end up being there to press buttons or do stuff then a wipe tower doesn't make sense to me, i would [...]

The wipe tower and the PS's filament swap algorithm makes a lot of sense even without a fully automated procedure. Because it allows to purge properly, properly form a filament tip (see the PS's ramming and skinnydip postprocessing script) and then set up a proper pressure in the hotend on a wipe tower before going to a model thus fully eliminating any printing artefacts like hanging string or initial pressure skip upon filament swaps. But that is just my experience and definitely is an offtopic. 

Anyway thanks for adding the setting. I'm sure a lot of people will be grateful to you for that, including me. Even if you still think that we are just a minority.

alfrix, 2023-09-05:
> set up a proper pressure in the hotend on a wipe tower before going to a model

i do a slow purge remove the filament manually and press resume while it's purging a little, perhaps i shall try a small tower when the model is small and don't have infill to purge itself

btw, i consider this a friendly interchange of thoughts don't be mad at me.

commit 2a842839 (2023-09-05): make extrude panel on pause optional

close #1113