KlipperScreen/KlipperScreen #1116: [FR] Use Weblate to facilitate translation

Daviteusz, 2023-09-08:
### Requested feature

I translate KlipperScreen using Weblate because it is a very convenient program for translators and is overall perfect for this project.

When the translation is completed, e.g. every 12 hours, the changes are automatically sent as a Pull Request. The entire operation rests on the shoulders of the weblate bot, translators don't need to know git usage to add new translations or update them. Weblate can also generate MO files.

The official Weblate server can host open-source projects for free.

### Additional information

[image]
[image]
[image]
[image]
[image]
[image]

alfrix, 2023-09-12:
sounds interesting, just created a test and invited you 
or we can use yours if you can add me as co-owner

Daviteusz, 2023-09-12:
I host my instance on a laptop and it is only available every few days, so it is better to use something that works without interruption. 

One good change can be made for better integration with the repository:
- Webhooks: https://docs.weblate.org/en/latest/admin/continuous.html#github-setup

I accepted your invite and I made some changes:
- I set pull request as the method for pushing out changes. It's also possible to push right out to the branch, but I don't feel that's a good idea.
- I also added `ignore-same` flags for **Input Shaper**, **Pressure Advance**, **Slicer**. In this case, translators will not be informed that the translation has not been changed.
- I also added an add-on that generates MO files after each PO change.

It is also possible to add screenshots to each string so that translators will know exactly what they are translating, but with that there is some work to be done.

It seems to me that this is enough to start with. Also, the names and descriptions of commits or pull requests can be changed, but overall the default is ok.

alfrix, 2023-10-15:
[image]

Daviteusz, 2023-11-09:
@alfrix I am currently forced to manually update the repository on Weblate. Could you add webhook in the repository settings? It's just few click :)

https://docs.weblate.org/en/latest/admin/continuous.html#automatically-receiving-changes-from-github

closed 2023-11-16 by Daviteusz

alfrix, 2023-11-16:
it seems that most people prefer this method of translation. Thank you [Daviteusz](https://github.com/Daviteusz)