KlipperScreen/KlipperScreen #1124: [BUG] <title>

BlackNet, 2023-09-25:
### What happened?

"fine-tuning" first run for a print, neither 1% or 2% is selected. Only select 2% if 1% is chosen already.  The ONLY way to choose 2% is first to press 1%.

### What did you expect to happen instead?

see above

### How to reproduce this bug?

fine-tuning on new print

### Log output

```shell
na
```

CIS94500, 2023-09-25:
Hello, I had the problem and I fixed it with this code, It may not be perfect but it works for me

> fine_tune.py

```
import logging
import re
import gi

gi.require_version("Gtk", "3.0")
from gi.repository import Gtk
from ks_includes.KlippyGcodes import KlippyGcodes
from ks_includes.screen_panel import ScreenPanel

class Panel(ScreenPanel):
    z_deltas = ["0.01", "0.05"]
    z_delta = z_deltas[-1]
    speed_deltas = ['5', '25']
    s_delta = speed_deltas[-1]
    extrude_deltas = ['1', '2']
    e_delta = extrude_deltas[-1]
    speed = extrusion = 100
    z_offset = 0.0

    def __init__(self, screen, title):
        super().__init__(screen, title)
        if self.ks_printer_cfg is not None:
            bs = self.ks_printer_cfg.get("z_babystep_values", "0.01, 0.05")
            if re.match(r'^[0-9,\.\s]+$', bs):
                bs = [str(i.strip()) for i in bs.split(',')]
                if 1 < len(bs) < 3:
                    self.z_deltas = bs
                    self.z_delta = float(self.z_deltas[-1])

        zgrid = Gtk.Grid()
        for j, i in enumerate(self.z_deltas):
            x = float(i)
            self.labels[f"zdelta{x}"] = self._gtk.Button(label=i)
            self.labels[f"zdelta{x}"].connect("clicked", self.change_percent_delta, "z_offset", x)
            ctx = self.labels[f"zdelta{x}"].get_style_context()
            if (self._screen.lang_ltr and j == 0) or (not self._screen.lang_ltr and j == len(self.z_deltas) - 1):
                ctx.add_class("distbutton_top")
            elif (not self._screen.lang_ltr and j == 0) or (self._screen.lang_ltr and j == len(self.z_deltas) - 1):
                ctx.add_class("distbutton_bottom")
            else:
                ctx.add_class("distbutton")
            if x == float(self.z_delta):
                ctx.add_class("distbutton_active")
            zgrid.attach(self.labels[f"zdelta{x}"], j, 0, 1, 1)

        spdgrid = Gtk.Grid()
        for j, i in enumerate(self.speed_deltas):
            x = int(i)
            self.labels[f"sdelta{x}"] = self._gtk.Button(label=f"{i}%")
            self.labels[f"sdelta{x}"].connect("clicked", self.change_percent_delta, "speed", x)
            ctx = self.labels[f"sdelta{x}"].get_style_context()
            if (self._screen.lang_ltr and j == 0) or (not self._screen.lang_ltr and j == len(self.speed_deltas) - 1):
                ctx.add_class("distbutton_top")
            elif (not self._screen.lang_ltr and j == 0) or (self._screen.lang_ltr and j == len(self.speed_deltas) - 1):
                ctx.add_class("distbutton_bottom")
            else:
                ctx.add_class("distbutton")
            if x == int(self.s_delta):
                ctx.add_class("distbutton_active")
            spdgrid.attach(self.labels[f"sdelta{x}"], j, 0, 1, 1)

        extgrid = Gtk.Grid()
        for j, i in enumerate(self.extrude_deltas):
            x = int(i)
            self.labels[f"edelta{x}"] = self._gtk.Button(label=f"{i}%")
[... 106 lines cut ...]
        if direction == "+":
            self.speed += int(self.s_delta)
        elif direction == "-":
            self.speed -= int(self.s_delta)
        elif direction == "reset":
            self.speed = 100

        self.speed = max(self.speed, 1)
        self.labels['speedfactor'].set_label(f"  {self.speed:3}%")
        self._screen._send_action(widget, "printer.gcode.script", {"script": KlippyGcodes.set_speed_rate(self.speed)})

    def change_percent_delta(self, widget, array, delta):
        logging.info(f"### Delta {array}_{delta}")
        widget.get_style_context().add_class("distbutton_active")
        if array == "z_offset" and delta != float(self.z_delta):
            self.labels[f"zdelta{self.z_delta}"].get_style_context().remove_class("distbutton_active")
            self.z_delta = delta
        elif array == "speed" and delta != int(self.s_delta):
            self.labels[f"sdelta{self.s_delta}"].get_style_context().remove_class("distbutton_active")
            self.s_delta = delta
        elif array == "extrude" and delta != int(self.e_delta):
            self.labels[f"edelta{self.e_delta}"].get_style_context().remove_class("distbutton_active")
            self.e_delta = delta

```

alfrix, 2023-10-03:
log not provided, and cannot reproduce the issue

closed 2023-10-03 by alfrix