KlipperScreen/KlipperScreen #1130: KlipperScreen just barely flashing on 3.5inch touch screen

f3Rd3r, 2023-09-28:
### What happened?

 I just setup a fresh install of latest Raspi OS lite with Klipper through Kiauh. When I am starting KlipperScreen, the 3.5inch touch screen stays black and flashes some parts of the KlipperScreen, when I touch the touch screen and goes black again. Drivers for the screen seem to be installed correctly. Anyone had this before? Cant find anything online about such a problem!?

When I just touch on the screen, it is able to preheat e.g. when I hit the place where a button is by accident. 

### What did you expect to happen instead?

KlipperScreen should show.

### How to reproduce this bug?

Using KlipperScreen on fresh RaspiOS lite with 3,5inch RPI touch screen (A)

### Log output

```shell
--------------------KlipperScreen Log Start--------------------
Git Version: v0.3.6-8-g685061f
2023-09-28 10:11:12,705 [screen.py:main()] - Python version: 3.9
2023-09-28 10:11:12,705 [screen.py:main()] - KlipperScreen version: v0.3.6-8-g685061f
2023-09-28 10:11:13,023 [config.py:get_config_file_location()] - Passed config (-c): /home/skid/KlipperScreen.conf
2023-09-28 10:11:13,026 [config.py:__init__()] - Config path location: /home/skid/printer_data/config/KlipperScreen.conf
2023-09-28 10:11:13,066 [config.py:log_config()] -  ===== Config File =====

=======================
2023-09-28 10:11:13,071 [config.py:__init__()] - ====== Saved Def ======

[graph Printer]
temperature_sensor mcu_temp = False
heater_bed = False

=======================
2023-09-28 10:11:13,073 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-09-28 10:11:13,303 [config.py:create_translations()] - Selected lang: None OS lang: en_GB
2023-09-28 10:11:13,304 [config.py:install_language()] - Using system lang
2023-09-28 10:11:13,305 [config.py:install_language()] - Using lang en
2023-09-28 10:11:13,335 [screen.py:__init__()] - Screen resolution: 480x320
2023-09-28 10:11:13,337 [KlippyGtk.py:__init__()] - Font size: 11.0 (medium)
2023-09-28 10:11:14,479 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2023-09-28 10:11:14,492 [screen.py:set_screenblanking_timeout()] - Using DPMS
2023-09-28 10:11:14,504 [screen.py:initial_connection()] - Default printer: None
2023-09-28 10:11:14,505 [screen.py:connect_printer()] - Connecting to printer: Printer
2023-09-28 10:11:14,505 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-09-28 10:11:14,615 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-09-28 10:11:14,632 [KlippyWebsocket.py:connect()] - Attempting to connect
2023-09-28 10:11:14,662 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-09-28 10:11:14,680 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2023-09-28 10:11:14,683 [KlippyWebsocket.py:connect()] - Starting websocket thread
2023-09-28 10:11:14,696 [_logging.py:info()] - Websocket connected
2023-09-28 10:11:14,696 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2023-09-28 10:11:14,740 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-09-28 10:11:14,757 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-09-28 10:11:14,759 [screen.py:init_printer()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'startup', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'authorization', 'octoprint_compat', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes'], 'warnings': ['[update_manager]: Failed to load extension crowsnest: Invalid path for option `path` in section [update_manager crowsnest]: Path `/home/skid/crowsnest` does not exist', '[update_manager]: Failed to load extension sonar: Invalid path for option `path` in section [update_manager sonar]: Path `/home/skid/sonar` does not exist', "Unparsed config option 'origin: https://github.com/mainsail-crew/crowsnest.git' detected in section [update_manager crowsnest].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'install_script: tools/pkglist.sh' detected in section [update_manager crowsnest].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'origin: https://github.com/mainsail-crew/sonar.git' detected in section [update_manager sonar].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'primary_branch: main' detected in section [update_manager sonar].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error.", "Unparsed config option 'install_script: tools/install.sh' detected in section [update_manager sonar].  This may be an option no longer available or could be the result of a module that failed to load.  In the future this will result in a startup error."], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-142-gee62d07', 'missing_klippy_requirements': [], 'api_version': [1, 3, 0], 'api_version_string': '1.3.0'}
2023-09-28 10:11:14,762 [screen.py:show_popup_message()] - warning: 
Moonraker warnings:
[update_manager]:
 Failed to load extension crowsnest:
 Invalid path for option `path` in section [update_manager crowsnest]:
 Path `/home/skid/crowsnest` does not exist
[update_manager]:
 Failed to load extension sonar:
[... 59 lines cut ...]
2023-09-28 10:11:18,947 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&configfile&display_status&extruder&fan&gcode_move&heater_bed&idle_timeout&pause_resume&print_stats&toolhead&virtual_sdcard&webhooks&motion_report&firmware_retraction&exclude_object&manual_probe&extruder&heater_bed&temperature_sensor raspberry_pi&temperature_sensor mcu_temp&fan&heater_fan hotend_fan&filament_switch_sensor filament_sensor&led LED_Light
2023-09-28 10:11:18,978 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2023-09-28 10:11:19,020 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_sensor raspberry_pi', 'temperature_sensor mcu_temp', 'extruder']
2023-09-28 10:11:19,038 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/config
2023-09-28 10:11:19,040 [screen.py:init_tempstore()] - Temperature store size: 1200
2023-09-28 10:11:19,041 [files.py:initialize()] - Gcodes path: /home/skid/printer_data/gcodes
2023-09-28 10:11:19,041 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2023-09-28 10:11:19,042 [screen.py:init_printer()] - Printer initialized
2023-09-28 10:11:19,043 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2023-09-28 10:11:19,044 [printer.py:change_state()] - Adding callback for state: ready
2023-09-28 10:11:19,633 [screen.py:_load_panel()] - Loading panel: main_menu
2023-09-28 10:11:19,815 [main_menu.py:__init__()] - ### Making MainMenu
2023-09-28 10:11:19,827 [main_menu.py:add_device()] - Adding device: extruder
2023-09-28 10:11:19,842 [main_menu.py:add_device()] - Adding device: heater_bed
2023-09-28 10:11:19,858 [main_menu.py:add_device()] - Adding device: temperature_sensor raspberry_pi
2023-09-28 10:11:19,868 [main_menu.py:add_device()] - Adding device: temperature_sensor mcu_temp
2023-09-28 10:11:19,955 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-09-28 10:11:43,983 [screen.py:_load_panel()] - Loading panel: extrude
2023-09-28 10:11:44,148 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > extrude
2023-09-28 10:11:44,152 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1
2023-09-28 10:11:44,156 [extrude.py:process_update()] - filament_switch_sensor filament_sensor: {'filament_detected': True, 'enabled': True}
2023-09-28 10:11:45,929 [screen.py:_send_action()] - printer.gcode.script: {'script': 'UNLOAD_FILAMENT SPEED=120'}
2023-09-28 10:11:46,135 [screen.py:show_popup_message()] - error: Extrude below minimum temp
2023-09-28 10:11:46,148 [screen.py:show_popup_message()] - error: Extrude below minimum temp
```

alfrix, 2023-09-28:
not a bug, klipperscreen is **working correctly** according to the log

> Drivers for the screen seem to be installed correctly

it doesn't seem like they are, but it also could be a cable or connection issue, or power issue (i have seen plenty of users with an inadequate power supply causing tons of issues)

https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Physical_Install/

alfrix, 2023-10-03:
closing as not an issue with KS itself

closed 2023-10-03 by alfrix