KlipperScreen/KlipperScreen #1131: [BUG] When trying to install KlipperScreen I get a python error

jrbenny35, 2023-09-28:
### What happened?

Using Kiuah to install KlipperScreen results in a python not found error.

### What did you expect to happen instead?

KlipperScreen to be installed

### How to reproduce this bug?

Starting with a fresh Ubuntu desktop install, I installed Kiauh and chose the install option and then selected KlipperScreen.

### Additional information:

It looks like the function ```check_requirements()``` is using ```python``` as the command, but I think it should be using ```python3``` like the following ```create_virtualenv()``` function.

### Log output

```shell
Checking Python version
/home/b4hand/KlipperScreen/scripts/KlipperScreen-install.sh: line 100: python: command not found
/home/b4hand/KlipperScreen/scripts/KlipperScreen-install.sh: line 101: python: command not found
Not supported

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#
```
```

jrbenny35, 2023-09-28:
I have also added an alias in my ```bashrc``` file to alias ```python3``` to ```python```

dog2bert, 2023-09-28:
I am trying to install KlipperScreen on my Neptune 4 Pro, but it is running python 2. Anyway to install still?

alfrix, 2023-09-28:
i have no taken into account old distros that have python 2 as default since python 2 has not been supported since 2020
this does mean  that you are probably using a buster based distro with python 3.7

keep in mind that **python 3.7 is out of support** 
and i will push KS requeriments to 3.9 when buster enters EOL mid-next year (bullseye has 3.9) 
3.8 will be out of support close to buster, it will be skipped

so anyone on legacy distros will not be able to update KS past july 2024

commit a3345f78 (2023-09-28): installer: call python3 close #1131

i have no taken into account old distros that have python 2 as default since python 2 has not been supported since 2020

keep in mind that python 3.7 is out of support
and i will push KS requeriments to 3.9 when buster enters EOL mid-next year (bullseye has 3.9)
3.8 will be out of support close to buster, it will be skipped

so anyone on legacy distros will not be able to update KS past july 2024
closed 2023-09-28 by alfrix