KlipperScreen/KlipperScreen #1133: [BUG] Loading panel: camera rise [error] ytdl_hook when using webrtc webcam

mhk-v0, 2023-10-01:
### What happened?

If you configure a Webcam using camera-streamer (webrtc) and access the webcam via Touchscreen i get an error:

IPs is altered to hide my local IP address!
```
2023-10-01 02:08:38,678 [camera.py:__init__()] - {'enabled': True, 'icon': 'mdiWebcam', 'aspect_ratio': '4:3', 'target_fps_idle': 5, 'name': 'Chamber', 'location': 'printer', 'service': 'webrtc-camerastreamer', 'target_fps': 30, 'stream_url': 'http://192.168.178.37:8080/webrtc', 'snapshot_url': 'http://10.1.1.2:8080/snapshot', 'flip_horizontal': False, 'flip_vertical': False, 'rotation': 0, 'source': 'config', 'extra_data': {}}
2023-10-01 02:08:38,716 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > camera
2023-10-01 02:08:38,716 [camera.py:play()] - video filters: rotate:0.0
2023-10-01 02:08:38,744 [camera.py:play()] - Camera URL: http://10.1.1.2:8080/webrtc
2023-10-01 02:08:41,063 [camera.py:log()] - [error] ytdl_hook: ERROR: Unsupported URL: http://192.168.178.37:8080/webrtc
2023-10-01 02:08:41,064 [screen.py:show_popup_message()] - error: ERROR: Unsupported URL: http://192.168.178.37:8080/webrtc
2023-10-01 02:08:41,084 [camera.py:log()] - [error] ytdl_hook: youtube-dl failed: unexpected error ocurred
2023-10-01 02:08:41,089 [screen.py:show_popup_message()] - error: youtube-dl failed: unexpected error ocurred
2023-10-01 02:08:41,094 [screen.py:_menu_go_back()] - #### Menu go back
```

### What did you expect to happen instead?

Same function as if you use "mjpegstreamer" as service in Moonraker.

### How to reproduce this bug?

Switching the webcam service in Moonraker from "mjpegstreamer" to "webrtc-camerastreamer" and editing the stream_url to match the config via moonraker.conf.

```
[webcam Chamber]
location: printer
#service: webrtc-camerastreamer
service: mjpegstreamer
stream_url: http://10.1.1.2:8080/webcam/?action=stream
#stream_url: http://10.1.1.2:8080/webrtc
target_fps: 30
target_fps_idle: 5
```

### Additional information:

Camera is working fine in Moonraker / Mainsail and can be accesed from any local device via URL in both configurations.

### Log output

```shell
2023-10-01 02:08:37,748 [screen.py:_load_panel()] - Loading panel: camera
2023-10-01 02:08:38,678 [camera.py:__init__()] - {'enabled': True, 'icon': 'mdiWebcam', 'aspect_ratio': '4:3', 'target_fps_idle': 5, 'name': 'Chamber', 'location': 'printer', 'service': 'webrtc-camerastreamer', 'target_fps': 30, 'stream_url': 'http://10.1.1.2:8080/webrtc', 'snapshot_url': 'http://10.1.1.2:8080/snapshot', 'flip_horizontal': False, 'flip_vertical': False, 'rotation': 0, 'source': 'config', 'extra_data': {}}
2023-10-01 02:08:38,716 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > camera
2023-10-01 02:08:38,716 [camera.py:play()] - video filters: rotate:0.0
2023-10-01 02:08:38,744 [camera.py:play()] - Camera URL: http://10.1.1.2:8080/webrtc
2023-10-01 02:08:41,063 [camera.py:log()] - [error] ytdl_hook: ERROR: Unsupported URL: http://10.1.1.2:8080/webrtc
2023-10-01 02:08:41,064 [screen.py:show_popup_message()] - error: ERROR: Unsupported URL: http://10.1.1.2:8080/webrtc
2023-10-01 02:08:41,084 [camera.py:log()] - [error] ytdl_hook: youtube-dl failed: unexpected error ocurred
2023-10-01 02:08:41,089 [screen.py:show_popup_message()] - error: youtube-dl failed: unexpected error ocurred
2023-10-01 02:08:41,094 [screen.py:_menu_go_back()] - #### Menu go back
2023-10-01 02:08:41,128 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more
```

alfrix, 2023-10-01:
closing as invalid, it's not a bug, you can open a Feature Request for webrtc, but the backend (mpv) doesn't seem to support it.

arekm, 2023-11-03:
Submitted: https://github.com/mpv-player/mpv/issues/12801

thijstriemstra, 2023-12-23:
I'm also seeing an error with `camera-streamer`:

```
Dec 23 15:42:30 klipperscreen KlipperScreen-start.sh[1016]: 2023-12-23 15:42:30,703 [screen.py:show_popup_message()] - error: ERROR: Unsupported URL: http://foobar.local/webcam/webrtc
Dec 23 15:42:30 klipperscreen KlipperScreen-start.sh[1016]: 2023-12-23 15:42:30,715 [camera.py:log()] - [error] ytdl_hook: youtube-dl failed: unexpected error occurred
Dec 23 15:42:30 klipperscreen KlipperScreen-start.sh[1016]: 2023-12-23 15:42:30,715 [camera.py:log()] - [error] cplayer: Failed to recognize file format.
Dec 23 15:42:30 klipperscreen KlipperScreen-start.sh[1016]: 2023-12-23 15:42:30,721 [screen.py:show_popup_message()] - error: Failed to recognize file format.
```

On the screen I see the `error: Failed to recognize file format.` error popup.

Applying @arekm's patch seems to work! So your assessment that it's an invalid bug report seems unfair @alfrix, it's something that can actually be fixed, or at least worked around, in the klipperscreen source code. It's better to have something than nothing, right?

Something like `if 'webrtc' in url` etc workaround until mpv-player doesn't sound like a terrible idea.  Thoughts?

alfrix, 2023-12-23:
if the protocol is not supported, then it should fail, and the user should use another protocol, we shouldn't be shadily changing urls in the background

arekm, 2023-12-23:
The problem is that browsers support that protocol just fine and only KlipperScreen (to be precise mpv under) is not. So now we are forced to "downgrade" everything to make mpv happy.

(and as I said - my patch is just a ugly local workaround, not a solution).

If webcam api would provide multiple stream urls to the same camera (since there are multiple streams available already) and KlipperScreen would choose the "best" one (or try one by one). That could be a possible solution.

alfrix, 2023-12-23:
> The problem is that browsers support that protocol just fine and only KlipperScreen (to be precise mpv under) is not. So now we are forced to "downgrade" everything to make mpv happy.
>
> (and as I [...]

perhaps it should try webrtc, and if fails then retry notifying the user like
```
'camera: webrtc is not supported by the backend trying stream'
```

closed 2023-12-23 by alfrix

thijstriemstra, 2023-12-24:
> perhaps it should try webrtc, and if fails then retry notifying the user like

thanks!

commit 4a928140 (2023-12-23): camera: workaround for webrtc if there is a stream close #1133