KlipperScreen/KlipperScreen #1137: [BUG] KlipperScreen not showing all enabled Macros

McFuzz89, 2023-10-03:
### What happened?

After one of the recent updates (currently running v0.3.6-14), the Macro menu stopped showing all available Macros. I keep all macro cfgs in a separate folder and observer that the three new Macros I created today show up, but some of the ones I created in the past no longer show up. 

As a test, I took one of the missing macros and renamed it, then copied its contents and added it to printer.cfg - after restarting klipper, that macro appeared in KlipperScreen, but the renamed one was still missing. 

### What did you expect to happen instead?

I expect all Macros in dedicated macro folder to show up. 

### How to reproduce this bug?

Unknown - I am not sure what led to this condition, but what I have done is I took one of the missing macros and renamed it, then copied its contents and added it to printer.cfg - after restarting klipper, that macro appeared in KlipperScreen, but the renamed one was still missing. 

### Additional information:

If you review the log output, in the wall of text you can see the missing macros - for example LOAD_FILAMENT and UNLOAD_FILAMENT...

### Log output

```shell
2023-10-02 17:48:27,391 [printer.py:change_state()] - Changing state from 'ready' to 'disconnected'
2023-10-02 17:48:27,391 [printer.py:change_state()] - Adding callback for state: disconnected
2023-10-02 17:48:27,392 [screen.py:state_disconnected()] - ### Going to disconnected
2023-10-02 17:48:27,397 [screen.py:show_panel()] - Reinitializing panel
2023-10-02 17:48:27,459 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2023-10-02 17:48:30,994 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-10-02 17:48:31,001 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2023-10-02 17:48:31,002 [screen.py:init_printer()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'startup', 'components': ['klippy_connection', 'application', 'websockets', 'internal_transport', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'secrets', 'template', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'history', 'update_manager', 'authorization'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 3, 'moonraker_version': 'v0.8.0-142-gee62d07', 'missing_klippy_requirements': [], 'api_version': [1, 3, 0], 'api_version_string': '1.3.0'}
2023-10-02 17:48:31,011 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2023-10-02 17:48:31,011 [printer.py:configure_cameras()] - Cameras: []
2023-10-02 17:48:31,020 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/info
2023-10-02 17:48:31,261 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2023-10-02 17:48:31,262 [screen.py:init_printer()] - {'configfile': {'save_config_pending_items': {}, 'warnings': [], 'config': {'gcode_macro CALIBRATE_Z_OFFSET': {'gcode': '\nG28\nPROBE_CALIBRATE'}, 'pause_resume': {}, 'gcode_macro FC_RESUME': {'gcode': '\n{% set E = printer["gcode_macro PAUSE"].extrude|float %}\nSET_IDLE_TIMEOUT TIMEOUT={ 10 * 60 }\nG1 E-{ E }\nG92 E0\nRESUME', 'description': 'Resume print after Filament Change'}, 'gcode_macro M600': {'gcode': '\n{% if printer.extruder.temperature < 180 %}\n{action_respond_info("Extruder temperature too low")}\n{% else %}\nPAUSE_MACRO\n_DISABLE_FS\nUNLOAD_FILAMENT\n{% endif %}', 'description': 'Starts process of Filament Change'}, 'gcode_macro BED_MESH_CALIBRATE_AND_SAVE': {'gcode': '\nG28\nBED_MESH_CALIBRATE\nG1 Z50\nSAVE_VARIABLE VARIABLE=print_count VALUE=1\nSAVE_CONFIG'}, 'gcode_macro COOL': {'gcode': '\nTURN_OFF_HEATERS', 'description': 'Turns off all heaters'}, 'gcode_macro END_PRINT': {'gcode': '\n{% set Y_PARK = printer.configfile.config["stepper_y"]["position_max"]|float*0.80 %}\n{% if printer.toolhead.position.z|float + 30 >= printer.configfile.config["stepper_z"]["position_max"]|float %}\nG90                  ;set absolute extrusion\nG1 Z{printer.configfile.config["stepper_z"]["position_max"]|float} F1000 ;Move tool\n{% else %}\nG91                  ;set relative extrusion\nG1 Z30 F1000         ;move Z axis up 30mm\nG90\n{% endif %}\nG1 X5 Y{Y_PARK}; present print\nM140 S0 ; turn off heatbed\nM104 S0 ; turn off temperature\nM107 ; turn off fan\nM84 X Y E ; disable motors'}, 'filament_switch_sensor RunoutSensor': {'runout_gcode': 'PAUSE', 'switch_pin': '^!PC15', 'pause_on_runout': 'True'}, 'gcode_macro LOAD_FILAMENT': {'gcode': '\n{% if printer.extruder.temperature < 180 %}\n{action_respond_info("Extruder temperature too low")}\n{% else %}\nSAVE_GCODE_STATE NAME=LOAD_state\nG91\nG1 E50 F{ 5 * 60 }\nG4 P{ 0.9 * 1000 }\nG1 E25.0 F{ 5 * 60 }\n_ENABLE_FS\nRESTORE_GCODE_STATE NAME=LOAD_state\n{% endif %}', 'description': 'Loads filament into the extruder'}, 'bed_mesh default': {'tension': '0.2', 'min_x': '26.0', 'min_y': '30.0', 'y_count': '10', 'mesh_y_pps': '2', 'x_count': '10', 'version': '1', 'algo': 'bicubic', 'mesh_x_pps': '2', 'max_y': '199.92', 'max_x': '199.97', 'points': '\n0.020000, 0.026250, 0.038750, 0.010000, -0.011250, 0.000000, -0.020000, -0.018750, 0.008750, 0.021250\n0.033750, 0.041250, 0.052500, 0.033750, 0.012500, 0.021250, 0.008750, 0.005000, 0.028750, 0.041250\n0.026250, 0.035000, 0.046250, 0.026250, 0.010000, 0.021250, 0.001250, 0.001250, 0.031250, 0.041250\n0.016250, 0.025000, 0.036250, 0.008750, -0.010000, 0.010000, -0.013750, -0.012500, 0.025000, 0.036250\n0.011250, 0.026250, 0.033750, 0.003750, -0.017500, 0.005000, -0.016250, -0.011250, 0.027500, 0.038750\n0.013750, 0.022500, 0.031250, 0.007500, -0.012500, 0.008750, -0.008750, -0.001250, 0.026250, 0.037500\n0.017500, 0.026250, 0.033750, 0.008750, -0.005000, 0.008750, -0.008750, 0.003750, 0.033750, 0.048750\n0.023750, 0.031250, 0.036250, 0.006250, -0.010000, 0.006250, -0.012500, 0.001250, 0.031250, 0.048750\n0.030000, 0.041250, 0.046250, 0.021250, 0.011250, 0.020000, 0.001250, 0.008750, 0.036250, 0.058750\n0.031250, 0.043750, 0.046250, 0.018750, 0.002500, 0.011250, -0.011250, -0.003750, 0.030000, 0.038750'}, 'virtual_sdcard': {'path': '~/printer_data/gcodes'}, 'gcode_macro _DISABLE_FS': {'gcode': '\nSET_FILAMENT_SENSOR SENSOR=RunoutSensor ENABLE=0', 'description': 'disables filament sensor that might trigger an M600 in the middle of a load/unload/M600'}, 'gcode_macro _ENABLE_FS': {'gcode': '\nSET_FILAMENT_SENSOR SENSOR=RunoutSensor ENABLE=1', 'description': 'enables filament sensor'}, 'gcode_macro CANCEL_PRINT': {'rename_existing': 'CANCEL_PRINT_BASE', 'description': 'Cancel the actual running print', 'gcode': '\nTURN_OFF_HEATERS\nCANCEL_PRINT_BASE'}, 'gcode_macro PAUSE_MACRO': {'gcode': '\nPAUSE\nSET_IDLE_TIMEOUT TIMEOUT={ 30 * 60 }', 'description': 'Pauses Print'}, 'gcode_macro G29': {'gcode': '\nBED_MESH_CALIBRATE_AND_SAVE'}, 'stepper_x': {'position_endstop': '-6', 'endstop_pin': '^!PA5', 'rotation_distance': '40', 'step_pin': 'PC2', 'position_min': '-6', 'microsteps': '16', 'homing_speed': '50', 'dir_pin': 'PB9', 'position_max': '242', 'enable_pin': '!PC3'}, 'heater_bed': {'control': 'pid', 'pid_kp': '72.739', 'sensor_pin': 'PC4', 'heater_pin': 'PA7', 'sensor_type': 'Generic 3950', 'pid_kd': '613.733', 'pid_ki': '2.155', 'min_temp': '0', 'max_temp': '100'}, 'delayed_gcode bed_mesh_init': {'gcode': '\nBED_MESH_PROFILE LOAD="default"', 'initial_duration': '.01'}, 'temperature_sensor Board_MCU': {'min_temp': '0', 'sensor_type': 'temperature_mcu', 'max_temp': '100'}, 'screws_tilt_adjust': {'screw1_name': 'LowerLeft', 'screw4_name': 'UpperRight', 'horizontal_move_z': '10', 'screw2_name': 'UpperLeft', 'speed': '150', 'screw_thread': 'CW-M3', 'screw2': '58, 240', 'screw3': '230, 70', 'screw1': '58, 70', 'screw3_name': 'LowerRight', 'screw4': '230, 240'}, 'gcode_macro START_PRINT': {'gcode': '\n{% set BED_TEMP = params.BED_TEMP|default(80)|float %}\n{% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(230)|float %}\nG90 ; use absolute coordinates\nM83 ; extruder relative mode\nM140 S{BED_TEMP} ; set final bed temp\nM104 S150 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling\nG4 S10 ; allow partial nozzle warmup\nG28 ; home all axis\nBED_MESH_PROFILE LOAD="default"\nG1 Z50 F240\nG1 X2 Y10 F3000\nM190 S{BED_TEMP}} ; wait for bed temp to s [... cut ...]
2023-10-02 17:48:31,695 [main_menu.py:__init__()] - ### Making MainMenu
2023-10-02 17:48:31,702 [main_menu.py:add_device()] - Adding device: extruder
2023-10-02 17:48:31,707 [main_menu.py:add_device()] - Adding device: heater_bed
2023-10-02 17:48:31,712 [main_menu.py:add_device()] - Adding device: temperature_sensor Board_MCU
2023-10-02 17:48:31,717 [main_menu.py:add_device()] - Adding device: temperature_sensor Raspberry_Pi
2023-10-02 17:48:31,747 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
```

alfrix, 2023-10-03:
not a bug
UNLOAD_FILAMENT and LOAD_FILAMENT are used in the extrude panel and for that reason they are removed from the macros
see https://github.com/KlipperScreen/KlipperScreen/issues/1104 for more info

closed 2023-10-03 by alfrix

McFuzz89, 2023-10-03:
> not a bug UNLOAD_FILAMENT and LOAD_FILAMENT are used in the extrude panel and for that reason they are removed from the macros see #1104 for more info

So to confirm - if I put my own load/unload Macros, the buttons in the Extruder panel will use those macro settings? What about other Macros that are not visible and dont have rename existing attribute?

commit 1b136a63 (2023-10-03): docs: clarify why some macros are no longer visible close #1137