KlipperScreen/KlipperScreen #1155: doc: upgrade instructions

thijstriemstra, 2023-10-25:
### What happened?

Would be nice to have some upgrade instructions in the documentation.

### What did you expect to happen instead?

Can only find installation instructions.

### How to reproduce this bug?

Try upgrading.

### Log output

```shell
-
```

thijstriemstra, 2023-10-28:
for now I created a bash script with:

```sh
#!/bin/bash

source ${HOME}/.KlipperScreen-env/bin/activate
pip --disable-pip-version-check install -r ~/KlipperScreen/scripts/KlipperScreen-requirements.txt
```

alfrix, 2023-10-31:
the recommendation is to configure moonraker as exaplained in the docs

https://klipperscreen.readthedocs.io/en/latest/Installation/#moonraker-configuration

after that is configured and moonraker is restarted you can use the system panel

[image]

thijstriemstra, 2023-10-31:
I run my KlipperScreen on a different Raspberry Pi that uses API to talk to a separate RPI in my network running Klipper/Moonraker. And in that situation, KlipperScreen isn't configured on the Klipper RPI, thus update function isn't available.

alfrix, 2023-11-01:
> I run my KlipperScreen on a different Raspberry Pi that uses API to talk to a separate RPI in my network running Klipper/Moonraker. And in that situation, KlipperScreen isn't configured on the Klipper RPI, thus update function isn't available.

makes sense, then your are missing the actual update:
```
cd ~/KlipperScreen
git pull
sudo service KlipperScreen restart
```

you can also use [KIAUH](https://github.com/dw-0/kiauh)

alfrix, 2023-11-04:
added with 524d049

closed 2023-11-04 by alfrix