KlipperScreen/KlipperScreen #1162: [BUG] <title>LOAD_FILAMENT  UNLOAD_FILAMENT error

skyaee, 2023-11-04:
### What happened?

Clicking LOAD_FILAMENT UNLOAD_FILAMENT on the screen displays the error:
MCU 'sht36v2' shutdown: Stepper too far in past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Configuration：

[gcode_macro LOAD_FILAMENT]
variable_load_distance:  50
variable_purge_distance:  25
gcode:
    {% set speed = params.SPEED|default(300) %}
    {% set max_velocity = printer.configfile.settings['extruder'].max_extrude_only_velocity  * 60 %}
    SAVE_GCODE_STATE NAME=load_state
    G91
    G92 E0
    G1 E{load_distance} F{max_velocity} # fast-load
    G1 E{purge_distance} F{speed} # purge
    RESTORE_GCODE_STATE NAME=load_state

[gcode_macro UNLOAD_FILAMENT]
variable_unload_distance:  50
variable_purge_distance:  25
gcode:
    {% set speed = params.SPEED|default(300) %}
    {% set max_velocity = printer.configfile.settings['extruder'].max_extrude_only_velocity  * 60 %}
    SAVE_GCODE_STATE NAME=unload_state
    G91
    G92 E0
    G1 E{purge_distance} F{speed} # purge
    G1 E-{unload_distance} F{max_velocity} # fast-unload
    RESTORE_GCODE_STATE NAME=unload_state

### What did you expect to happen instead?

Click LOAD_FILAMENT UNLOAD_FILAMENT on the screen

### How to reproduce this bug?

Click LOAD_FILAMENT UNLOAD_FILAMENT on the screen

### Log output

```shell
2023-11-04 20:28:38,088 [main_menu.py:add_device()] - Adding device: temperature_sensor Motor_X
2023-11-04 20:28:38,092 [main_menu.py:add_device()] - Adding device: temperature_sensor Motor_Y
2023-11-04 20:28:38,097 [main_menu.py:add_device()] - Adding device: temperature_sensor 挤出机电机
2023-11-04 20:28:38,102 [main_menu.py:add_device()] - Adding device: temperature_sensor BED_2
2023-11-04 20:28:38,107 [main_menu.py:add_device()] - Adding device: temperature_sensor Pi
2023-11-04 20:28:38,112 [main_menu.py:add_device()] - Adding device: temperature_fan 树莓派4B
2023-11-04 20:28:38,115 [main_menu.py:add_device()] - Adding device: temperature_fan 下位机散热
2023-11-04 20:28:38,119 [main_menu.py:add_device()] - Adding device: temperature_fan 电器仓左风扇
2023-11-04 20:28:38,123 [main_menu.py:add_device()] - Adding device: temperature_fan 排气扇Z
2023-11-04 20:28:38,155 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
```

alfrix, 2023-11-04:
this is not a KlipperScreen bug, you have overloaded your mcu with steps.

the proper way to fix this is to configure `max_extrude_only_velocity` in your printer.cfg
https://www.klipper3d.org/Config_Reference.html#extruder

if it has been configured, then use a lower value

if you would have put the complete log, i would been able to give you a starting value

closed 2023-11-04 by alfrix

skyaee, 2024-01-07:
> 这不是 KlipperScreen 错误，您的 MCU 的步骤已超载。
> 
> 解决此问题的正确方法是`max_extrude_only_velocity`在 Printer.cfg 中进行配置 https://www.klipper3d.org/Config_Reference.html#extruder
> 
> 如果已配置，则使用较低的值
> 
> 如果您输入完整的日志，我就能给您一个起始值

[Uploading klippy.zip…]()