KlipperScreen/KlipperScreen #1184: [BUG] <title>When I change the wifi dongle it gets fixed to eth0.

NNGB-kor, 2023-11-27:
### What happened?

If you remove the wifi dongle set during armbian first run and use another wifi dongle, it will be fixed to eth0 and the wifi list will not be updated.
And the wifi dongle is not recognized as wlan0, but with the name wlxxxxxxxxxxx.
I think this has changed due to Linux policy.
For example, if the name of the existing WiFi dongle was wlx123456789 and the name of the new WiFi dongle was wlx987654321.
Klipper Screen's WiFi stops working until wlx12345679 is recognized.
If device wlxxxxxx exists, can't I do a fresh search for this?

### What did you expect to happen instead?

I thought that even if I change the wifi dongle, it will be detected as wlan0 and the wifi list will be updated and connection will be possible.

### How to reproduce this bug?

This occurs when you install armbian jammy 6.6 kernel version and change the wifi dongle.

### Log output

```shell
null
```

NNGB-kor, 2023-11-27:
sudo ln -s /dev/null /etc/udev/rules.d/80-net-setup-link.rules
I changed the interface name to wlan0 using the command.
Luckily wifi works.
However, this operation requires a keyboard and is impossible for people who do not know Linux.

alfrix, 2023-11-27:
switching devices during runtime is not supported at the moment