KlipperScreen/KlipperScreen #1195: [FR] Support more than 5 extruders.

viesturz, 2023-12-09:
### Requested feature

The extruder menu will only show up to 5 extruders. I'm running 6 and that makes the last one inaccessible via the screen.

thijstriemstra, 2024-01-14:
> The extruder menu will only show up to 5 extruders.

Can you add a screenshot?

viesturz, 2024-01-15:
[image]
Sure.
[image]

I tried swiping sideways to get to E5 but that does not work as well.

PR #1249 (2024-01-26): extrude: add an extruder selector for machines with more than 5 extruders
makes the extrude panel support more extruders (unlimited, icons from 0-11)

close #1195
commit 5b7da9a9 (2024-01-26): extrude: add an extruder selector for machines with more than 5 extruders (#1249)

makes the extrude panel support more extruders (unlimited, but icons from 0-11)
close #1195
closed 2024-01-26 by alfrix