KlipperScreen/KlipperScreen #1196: [BUG] KlipperScreen  update caused pi camera to stop working

C8forT, 2023-12-10:
### What happened?

I have had KlipperScreen installed with Fluidd for quite some time, and my pi camera worked as expected.

However, I updated KlipperScreen via the Fluidd UI and my pi camera stopped working.  (I spent much time trouble shooting before determining that the KlipperScreen update was the culprit).

I then installed an SD card I had with Mainsail, which did NOT have KlipperScreen installed.  My pi camera again worked as expected.  I then installed KlipperScreen on that system, and my pi again camera stopped working.

I wiped my Fluidd SSD and installed the latest version of Raspian (based on Bookworm), and re-installed everything.  Pi camera still does not work with KlipperScreen installed.  I also tried installing Crowsnest, but that did not fix the issue.

### What did you expect to happen instead?

When KlipperScreen is updated or installed, my pi camera continues to work.

### How to reproduce this bug?

Update or install KlipperScreen with Fluidd or Mainsail on a Raspberry pi that has a pi camera connected.

### Additional information:

Strangely, uninstalling KlipperScreen did not get the pi camera working again.  The only thing that worked was wiping the SD card and installing everything except KlipperScreen.  Then my pi camera works.

### Log output

```shell
/home/tforsythe/Downloads/KlipperScreen.log
```

alfrix, 2023-12-12:
the installer just installs the dependencies klipperscreen needs to run and no changes to the system are done beyond the dependency install, so it's very possible that i would not be able to fix since it's really not a klipperscreen bug

fot that reason it sounds like a dependency conflict issue, it would be needed a clean install and installing packages from the installer one by one to verify what 'breaks' the camera

alfrix, 2023-12-27:
This is not a bug of KlipperScreen, a bug of KlipperScreen is something related to the application itself, 
this is just an issue in your OS related to the HW

NerdyGriffin, 2024-06-10:
1) Make sure that the camera was not automatically disabled by one of the updates, either by using `raspi-config` or manually editing the `/boot/config.txt` file.

2) There is a known compatibility issue between KlipperScreen (or one of its dependencies) and the new camera stack "libcamera".
The Camera Module v3 is not compatible with the legacy camera stack, so (as of the time I am writing this) it requires libcamera to function.
I do not recall the root cause of why/how KlipperScreen breaks the libcamera stack, it has been months since I last tried to look into this issue. I don't know what model of Pi Camera you are using, but if it is a v3 then this is likely the cause.

ploucandco, 2024-08-06:
I am using an RPI zero 2W with a camera module 3 and a waveshare 4inches SPI touchscreen. Always get a very very low FPS (1-3 when printing) on the camera but the CPU seems not to be too much loaded (less than 20%). Any reason that could be related to this issue?

matte-oss, 2024-10-03:
I have the same problem, how do i fix this?