KlipperScreen/KlipperScreen #1198: [BUG] <After clicking the move button in motion control, the command is executed but the button status changes to waiting.>

zhengxiong404, 2023-12-12:
### What happened?

When the button is clicked, it executes the action correctly, but the button remains in a waiting state without ending. Exiting and re-entering the motion control interface restores the button.

### What did you expect to happen instead?

Click the move button, execute the movement. During axis movement, the button is in a waiting state. After axis movement ends, the button restores.

### How to reproduce this bug?

Click the move button

### Additional information:

[image]

[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/13645929/KlipperScreen.log)

### Log output

```shell
2023-12-12 16:52:47,779 [main_menu.py:__init__()] - ### Making MainMenu
2023-12-12 16:52:47,840 [main_menu.py:add_device()] - Adding device: extruder
2023-12-12 16:52:47,882 [main_menu.py:add_device()] - Adding device: heater_bed
2023-12-12 16:52:48,119 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-12-12 16:55:23,089 [screen.py:_load_panel()] - Loading panel: move
2023-12-12 16:55:23,637 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > move
2023-12-12 16:55:24,884 [screen.py:_send_action()] - printer.gcode.script: {'script': 'G91\nG0 X-25 F3000'}
2023-12-12 16:55:24,904 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
2023-12-12 16:55:27,592 [screen.py:_send_action()] - printer.gcode.script: {'script': 'G91\nG0 X+25 F3000'}
2023-12-12 16:55:27,610 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
2023-12-12 16:55:29,583 [screen.py:_send_action()] - printer.gcode.script: {'script': 'G91\nG0 Y-25 F3000'}
2023-12-12 16:55:29,602 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
2023-12-12 16:55:30,977 [screen.py:_send_action()] - printer.gcode.script: {'script': 'G91\nG0 Y+25 F3000'}
2023-12-12 16:55:31,012 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
2023-12-12 17:01:20,864 [screen.py:_menu_go_back()] - #### Menu go back
2023-12-12 17:01:21,010 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2023-12-12 17:01:21,617 [screen.py:show_popup_message()] - error: Move out of range: 80.000 210.000 10.000 [0.000]
2023-12-12 17:01:28,160 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > move
2023-12-12 17:01:28,392 [KlippyGtk.py:Dialog()] - Showing dialog KlipperScreen (width=1280, height=720)
2023-12-12 17:01:29,973 [KlippyGtk.py:remove_dialog()] - Removing Dialog
```

alfrix, 2023-12-12:
you seem to be running as as root?
```
 [config.py:__init__()] - Config path location: /root/KlipperScreen.conf
```
and
```
[functions.py:get_software_version()] - Error getting git version: b"fatal: detected dubious ownership in repository at '/home/umeko/KlipperScreen'\nTo add an exception for this directory, call:\n\n\tgit config --global --add safe.directory /home/umeko/KlipperScreen\n"
```
for that your bug report is not valid, when you resolve this issues, if it happens on a valid hash version of the main repo without themes you can continue on #1087

closed 2023-12-12 by alfrix

zhengxiong404, 2023-12-12:
Thank you for your assistance. Due to the use of an old Android phone as the host device, KlipperScreen cannot modify WiFi settings unless run with root privileges.