KlipperScreen/KlipperScreen #1200: Title cut off on dashboard

thijstriemstra, 2023-12-12:
### What happened?

The klipperscreen dashboard cuts off titles that otherwise should fit on the screen.

### What did you expect to happen instead?

Display full title.

### How to reproduce this bug?

add a thermal device with a title that contains 9 characters, eg CAN EXT 1. It will be displayed as “CAN EX…” even though “CAN EXT 1” fits in that part of the GUI.

[image]

### Additional information:

Using RPI3b with hyperpixel 4 rectangular screen.

### Log output

```shell
nothing in the log related to this issue.
```

alfrix, 2023-12-12:
the temp column is very wide because it has to contain labels such as 250 / 250 
and would be worse if you add the power too
and here you combined all the factors: scrollbar, very large fonts, small resolution

i do not not considered this a bug, it could be an feature request of variable column width, which idk if it's achievable on pygtk

> nothing in the log related to this issue.

and the logfile for a bugreport is always required as stated in the issue template

alfrix, 2023-12-12:
i think it can be enhanced via a smaller width if no target is active, i will try, but ultimately it will be cut as soon as more info is in the temp column

thijstriemstra, 2023-12-12:
> it will be cut as soon as more info is in the temp column

I think that's a really good workaround.

thijstriemstra, 2023-12-12:
Another issue that can be seen in the screenshot is the scrollbar. It would be really nice to fit 4 items in the list without having to display a scrollbar. I think the rows can be shrunk a little more in order to hide the scrollbar but I'll leave it up to you.

commit 02ba0f95 (2023-12-23): heater grid: smaller temp column if power is not displaying close #1200
closed 2023-12-23 by alfrix