KlipperScreen/KlipperScreen #1201: [BUG] Cannot shutdown host

thijstriemstra, 2023-12-12:
### What happened?

When trying to shutdown the host system for the GUI, it shows the following error in the log:

```
Dec 12 14:08:53 klipperscreen systemctl[1062]: Call to PowerOff failed: Interactive authentication required.
```

I'm using RPI OS 12 (Bookworm) and `/etc/X11/Xwrapper.config` contains:

```
# Xwrapper.config (Debian X Window System server wrapper configuration file)
#
# This file was generated by the post-installation script of the
# xserver-xorg-legacy package using values from the debconf database.
#
# See the Xwrapper.config(5) manual page for more information.
#
# This file is automatically updated on upgrades of the xserver-xorg-legacy
# package *only* if it has not been modified since the last upgrade of that
# package.
#
# If you have edited this file but would like it to be automatically updated
# again, run the following command as root:
#   dpkg-reconfigure xserver-xorg-legacy
needs_root_rights=yes
allowed_users=anybody
```

Also note that this KlipperScreen runs on a separate RPI, and Moonraker/Klipper run on another RPI.

### What did you expect to happen instead?

No error, and host shutdown.

### How to reproduce this bug?

Go to system -> shutdown -> host system

### Log output

```shell
● KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; preset: enabled)
     Active: active (running) since Tue 2023-12-12 01:20:15 CET; 12h ago
   Main PID: 497 (KlipperScreen-s)
      Tasks: 7 (limit: 765)
        CPU: 2min 42.267s
     CGroup: /system.slice/KlipperScreen.service
             ├─497 /bin/bash /home/thijs/KlipperScreen/scripts/KlipperScreen-start.sh
             ├─512 /usr/bin/xinit /home/thijs/.KlipperScreen-env/bin/python /home/thijs/KlipperScreen/screen.py
             ├─549 /usr/lib/xorg/Xorg :0
             └─747 /home/thijs/.KlipperScreen-env/bin/python /home/thijs/KlipperScreen/screen.py

Dec 12 14:08:21 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:21,668 [system.py:_already_updated()] - klipper v0.12.0-31
Dec 12 14:08:21 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:21,672 [system.py:_already_updated()] - led_effect v0.0.10-1
Dec 12 14:08:21 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:21,675 [system.py:_already_updated()] - moonraker v0.8.0-206
Dec 12 14:08:24 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:24,268 [KlippyGtk.py:Dialog()] - Showing dialog Shutdown (width=83>
Dec 12 14:08:31 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:31,205 [KlippyGtk.py:remove_dialog()] - Removing Dialog
Dec 12 14:08:31 klipperscreen systemctl[1041]: Call to PowerOff failed: Interactive authentication required.
Dec 12 14:08:51 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:51,340 [KlippyGtk.py:Dialog()] - Showing dialog Shutdown (width=83>
Dec 12 14:08:53 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:08:53,667 [KlippyGtk.py:remove_dialog()] - Removing Dialog
Dec 12 14:08:53 klipperscreen systemctl[1062]: Call to PowerOff failed: Interactive authentication required.
Dec 12 14:13:53 klipperscreen KlipperScreen-start.sh[747]: 2023-12-12 14:13:53,924 [screen.py:show_screensaver()] - Showing Screensaver
```
```

thijstriemstra, 2023-12-12:
After some research it looks like it might need something like https://github.com/Arksine/moonraker/blob/master/scripts/set-policykit-rules.sh?

But that's a lot of code though, hopefully it can be brought down to a few lines of code..

leoshusar, 2023-12-14:
Maybe related, but connecting to WiFi from KlipperScreen is not working for me either.
When I rescan, I get this in log:
```
2023-12-14 09:21:03,061 [wifi_nm.py:rescan()] - Error during rescan org.freedesktop.NetworkManager.PermissionDenied: org.freedesktop.NetworkManager.wifi.scan request failed: not authorized
```

alfrix, 2023-12-15:
i am aware of this i did some work some months ago in a branch https://github.com/alfrix/KlipperScreen/tree/polkit

the wifi thing is also known https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Network/

it's just another permission to add to the installer in the polkit branch
org.freedesktop.NetworkManager.wifi.scan

the branch it's still untested

alfrix, 2023-12-17:
#1205  should fix this

closed 2023-12-23 by alfrix

thijstriemstra, 2024-07-25:
Just FYI, I ran into this issue again and turns out the error is still thrown when Moonraker is down:

```
Jul 25 20:16:45 ender KlipperScreen-start.sh[815]: 20240725 20:16:45,517 [KlippyRest.py:_do_request] - host=127.0.0.1, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
Jul 25 20:16:45 ender KlipperScreen-start.sh[815]: 20240725 20:16:45,517 [screen.py:connect_to_moonraker] - Cannot get server info
Jul 25 20:16:49 ender KlipperScreen-start.sh[815]: 20240725 20:16:49,505 [screen.py:connect_to_moonraker] - Stopping Retries
Jul 25 20:17:31 ender KlipperScreen-start.sh[815]: 20240725 20:17:31,874 [KlippyGtk.py:Dialog] - Showing dialog Restart (width=800, height=480)
Jul 25 20:17:34 ender KlipperScreen-start.sh[815]: 20240725 20:17:34,204 [KlippyGtk.py:remove_dialog] - Removing Dialog
Jul 25 20:17:34 ender systemctl[969]: Call to Reboot failed: Interactive authentication required.
Jul 25 20:31:59 ender KlipperScreen-start.sh[815]: 20240725 20:31:59,220 [KlippyGtk.py:Dialog] - Showing dialog Restart (width=800, height=480)
Jul 25 20:32:00 ender KlipperScreen-start.sh[815]: 20240725 20:32:00,435 [KlippyGtk.py:remove_dialog] - Removing Dialog
Jul 25 20:32:00 ender systemctl[1006]: Call to Reboot failed: Interactive authentication required.
Jul 25 20:37:00 ender KlipperScreen-start.sh[815]: 20240725 20:37:00,728 [screen.py:show_screensaver] - Showing Screensaver
Jul 25 20:39:48 ender KlipperScreen-start.sh[815]: 20240725 20:39:48,718 [screen.py:close_screensaver] - Closing Screensaver
Jul 25 20:39:48 ender KlipperScreen-start.sh[815]: 20240725 20:39:48,741 [screen.py:wake_screen] - Screen wake up
Jul 25 20:44:48 ender KlipperScreen-start.sh[815]: 20240725 20:44:48,728 [screen.py:show_screensaver] - Showing Screensaver
Jul 25 20:59:20 ender KlipperScreen-start.sh[815]: 20240725 20:59:20,385 [screen.py:close_screensaver] - Closing Screensaver
Jul 25 20:59:20 ender KlipperScreen-start.sh[815]: 20240725 20:59:20,406 [screen.py:wake_screen] - Screen wake up
Jul 25 20:59:24 ender KlipperScreen-start.sh[815]: 20240725 20:59:24,894 [KlippyGtk.py:Dialog] - Showing dialog Shutdown (width=800, height=480)
Jul 25 20:59:26 ender KlipperScreen-start.sh[815]: 20240725 20:59:26,895 [KlippyGtk.py:remove_dialog] - Removing Dialog
Jul 25 20:59:27 ender systemctl[2328]: Call to PowerOff failed: Interactive authentication required.
```

which is probably a moonraker issue but it was confusing nonetheless (the error message doesn't indicate what program the error is coming from).

commit be972d50 (2023-12-23): Polkit rules to fix #1201 and improve network permissions avoiding the nm workaround (#1205)

* installer: use polkit for permissions

* update dependencies

* docs update