KlipperScreen/KlipperScreen #1215: [BUG] Unable to install (iptables)

lin-ycv, 2023-12-27:
### What happened?

Trying to install a fresh instance to klipper and klipperscreen on Flsun V400, following these instructions [here](https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad/wiki)
I've done this before, and it installs fine, but this time it's not working when installing klipperscreen
as seen [here](https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad/discussions/151) and [here](https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad/issues/29), multiple people are having the same issue
when installing klipperscreen, it gets stuck on `setting up iptables (1.8.4-3ubuntu2.1)` and soft bricks the speeder pad.

I've tried using the custom build of klipperscreen from @Guilouz , i've also tried installing the official build with `kiauh`, neither can install correctly.

### What did you expect to happen instead?

for the install of klipperscreen to proceed normally

### How to reproduce this bug?

do a clean install of klipper and klipperscreen

### Additional information:

Printer: Flsun V400

### Log output

```shell
There are no logs because the install bricks the speeder pad, unable to get the logs.
```

Maggan0, 2023-12-27:
Same here. After the Klipper screen update my FLSUN V400 Speeder Pad not loading the Klipper screen, had no WiFi Connection and SSH connection was not possible. So I reinstalled Klipper, but the installation of Klipper screen ends up the same like the update. 
I tried it three times.

giuspo, 2023-12-27:
Hi,
I'm noting the same problem, but I don't know if it is iptables. Because the second time, I performed the Klipperscreen script, stopped elsewhere, but it seems after "Network -manager".

Wrathlord12, 2023-12-27:
idem probleme a l'installation bloquer a iptables 
[image]

alfrix, 2023-12-27:
removing `network-manager` from system-dependencies.json will probably fix this

This is not a bug of KlipperScreen, a bug of KlipperScreen is something related to the application itself, this is just an issue in your OS related to the HW

alfrix, 2023-12-27:
idk how you are connecting to the device, but if it's via network, then it doesn't "brick the pad", you need local access to re-establish the network

Wrathlord12, 2023-12-27:
le pad est connecté en wifi via le réseau , l'accès ssh est impossible une fois l'erreur apparu

giuspo, 2023-12-27:
> le pad est connecté en wifi via le réseau , l'accès ssh est impossible une fois l'erreur apparu

Yes, I have the same issue, Even my router no longer sees the speeder pad.

giuspo, 2023-12-27:
Update...
I removed "network-manager" from the "OPTIONAL" in the script "KlipperScreen-Flsun-Speeder-Pad/scripts/KlipperScreen-install.sh" and the installation of Klipperscreen is finished:

[image]

Maggan0, 2023-12-27:
I am not an expert. Don't know how to do this. Do you have an instruction for me? ☺️

Wrathlord12, 2023-12-27:
> Je ne suis pas un expert. Je ne sais pas comment faire ça. Avez-vous une instruction pour moi?☺️

idem je suis preneur de la methode merci

giuspo, 2023-12-27:
In Linux I use "nano" to edit text file.
So, in console edit
sudo nano KlipperScreen/scripts/KlipperScreen-install.sh and after put your password
[image]

In nano editor find the OPTIONAL row:
[image]

and remove "network-manager" (really I copied the entire row and paste without network-manager)

after press "CTRL" and "X" to save:
[image]

so, execute the script "KlipperScreen-Flsun-Speeder-Pad/scripts/KlipperScreen-install.sh" (like in the wiki) and so on... :-)

giuspo, 2023-12-27:
So, I don't know if the "network-manager"" is important 🤷‍♀️

lin-ycv, 2023-12-27:
> So, I don't know if the "network-manager"" is important 🤷‍♀️
from the git history, `network-manager` was only added 4 days ago (to fix a network issue?)
since I have no issues with the previous installs that didn't have it, it's probably fine

closed 2023-12-27 by lin-ycv

alfrix, 2023-12-27:
> So, I don't know if the "network-manager"" is important 🤷‍♀️

it's necessary for the network panel, the wpa_cli method will be deprecated in 6 months along with 3.7 and debian-buster support

even if you lose connection while installing, if ks starts after the install, then you should be able to reconnect using the network panel, keep in mind that it will probably take a while to finish install after the connection is lost

lin-ycv, 2023-12-27:
> it's necessary for the network panel, the wpa_cli method will be deprecated in 6 months along with 3.7 and debian-buster support
>
> even if you lose connection while installing, if ks starts after the [...]

is there a rough estimate of "a while"? on my first and second install I waited for over an hour, it never connected back onto my wifi. That's why I was under the impression it was soft-bricked.

alfrix, 2023-12-27:
> it never connected back onto my wifi. 

it will not connect automatically, as said you need to reconnect using ks or using a keyboard and local tty (idk if that is possible in this device)

giuspo, 2023-12-27:
> it's necessary for the network panel, the wpa_cli method will be deprecated in 6 months along with 3.7 and debian-buster support
>
> even if you lose connection while installing, if ks starts after the [...]

I noticed that Klipperscreen is also stuck. So, you can not go into the network panel.
Just try to redo the procedure and you will notice the problem that I am having like the others.

giuspo, 2023-12-27:
> > it never connected back onto my wifi.
> 
> it will not connect automatically, as said you need to reconnect using ks or using a keyboard and local tty (idk if that is possible in this device)

I tried also with a keyboard, but I could not get a console in order to verify the Network connection. Speeder Pad seems stucked!

alfrix, 2023-12-27:
> I noticed that Klipperscreen is also stuck. So, you can not go into the network panel. Just try to redo the procedure and you will notice the problem that I am having like the others.

i must remind you that i don't have this device, and the nuances are unknown to me, i only have a couple of Raspberry Pi
if your device has a bug in network-manager, fixing it's outside of the scope of this repo, you can blacklist the package to prevent installs via apt-pinning or apt-mark

giuspo, 2023-12-27:
> i must remind you that i don't have this device, and the nuances are unknown to me, i only have a couple of Raspberry Pi if your device has a bug in network-manager, fixing it's outside of the scope [...]

I think I will do as you say,, I blacklist the package.
Thanks for your support.