KlipperScreen/KlipperScreen #1226: [BUG] Left side of screen cut off, touches not impacting correct locations.

stumcewing, 2024-01-06:
### What happened?

When booting KlipperScreen the logo loads in one corner and then when the screen loads the menus are all cut off on the left side, temperature settings are not properly displayed and the touch buttons do not accurately represent the locations.

### What did you expect to happen instead?

That klipperscreen would load with the menus in the correct positions as screen resolution is set correctly within the O/S for the BTT-HDMI5 (800x480). It is set to that same resolution in the BoardEnv.txt using the extraargs=video=HDMI-A-1:800x480-24@60 command. Moreover KlipperScreen log shows the resolution being detected as 800x480. 

### How to reproduce this bug?

Load Klipperscreen, it occurs every single time

### Additional information:

It's a BTT Manta M8P with a CB1 running the BTT Klipper build, which comes with Klipperscreen already installed. Klipperscreen is up to date according to Klipper: (v0.3.8-26-gd92ebac8)
[image]
[image]
[image]
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/13848082/KlipperScreen.log)

### Log output

```shell
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/13848082/KlipperScreen.log)
```

alfrix, 2024-01-06:
if that would have been the klipperscreen miss-behaving in the log it would have appeared as `update_size Size changed` , and it never looks stretched out like that

to me that looks like an improper resolution set, (resolution is much more than just the size in pixels)
maybe you forgot to do this:
https://bigtreetech.github.io/docs/HDMI5.html#hdmi-display-output

stumcewing, 2024-01-06:
That specifically is for running it with an RPi, I'm using it with the CB1 module from BTT, it's the Pi replacement module that they created for their boards. In that case you have to use the BoardEnv.txt folder as it doesn't have a config.txt like an RPi would. in that file I have the argument extraargs=video=HDMI-A-1:800x480-24@60 uncommented to force the setting of the resolution to 800x480, which I do see reflected in the logs.
 
 That whole process is detailed here: https://github.com/bigtreetech/CB1

alfrix, 2024-01-06:
i'm 100% sure from what you are showing and the log output than this is not a bug of klipperscreen, it's an issue with the configuration or the display drivers

closed 2024-01-06 by alfrix

alfrix, 2024-01-06:
You can try a different version of the image or an older one, or directly contact btt support