KlipperScreen/KlipperScreen #1235: Bed mesh calibrate button greyed out

thijstriemstra, 2024-01-12:
### What happened?

when i press calibrate button for bed mesh it becomes greyed out and hardly readable, even though it shows an important status message. 

[image]

### What did you expect to happen instead?

i’d rather see the status message in clear colors in center of screen (and temporary hide the calibrate button).

### How to reproduce this bug?

calibrate bed mesh

### Log output

```shell
no log file, posting from mobile phone
```

thijstriemstra, 2024-01-12:
Also, the text in the center "No mesh has been loaded" cannot be found in the translations..?

alfrix, 2024-01-13:
> Also, the text in the center "No mesh has been loaded" cannot be found in the translations..?

[8746456](https://github.com/KlipperScreen/KlipperScreen/commit/874645614f33cde1e54fa2f50e53f570041ab54d)